richonguzman/LoRa_APRS_Tracker #111: Tracker works but shows "Nobattery Connected"

wernizh, 2024-05-23:
My tracker does work but has a small issue with the displayed battery voltage
It's a Lilygo T-Beam V1.1 LORA32 it does have an onboard 18650 battery.

a) when running from the onboard 18650 battery it shows:
    on the display:  "No Battery Connected"
    on aprs.fi:     Bat=0.00V (%)
b) when connected to a power supply:  (charging the onboard 18650 battery)
    on the display:  0.29V   0%
    on aprs.fi:      Bat=0.00V (%)

In vscode I did choose ttgo-t-beam-v1_SX1268  not exactly the model I have but the closed one to choose from.
   
It's not a bit issue but if it could be resolved it would be usefull. (to know the battery voltage when walking around).

Thanks a lot,

Werner

richonguzman, 2024-05-23:
send me a picture of your board as the solution is pretty easy

wernizh, 2024-05-23:
[image]

wernizh, 2024-05-23:
[image]

richonguzman, 2024-05-23:
use the ENV:

"ttgo-t-beam-v1" it should work right away

wernizh, 2024-05-23:
Thanks a lot Ricardo for your fast feedback. 

I tried 3 times.  Also with beginning from scratch but still the same problem.

Hard resetting via RTS pin...
========================== [SUCCESS] Took 26.63 seconds ==============================================

Environment     Status    Duration
--------------  --------  ------------
ttgo-t-beam-v1  SUCCESS   00:00:26.632
========================= 1 succeeded in 00:00:26.632 =============================================

richonguzman, 2024-06-02:
1) check again the ENVIRONMENT selection as this is MANDATORY to select it right. You should select "ttgo-t-beam-v1" and upload firmware and then go into the upload filesystem.

2) it this still continues, sometimes AXP192 and AXP2101 power management chips needs a reset (so you take out usb and also 18650 battery and then put the battery again to do the power management chip reset)

wernizh, 2024-06-02:
Thanks a lot Ricardo

It does work now!

I did download the latest version of the tracker (2024-05-30) and set the environment to "ttgo-t-beam-v1".
(Before I had the environment set to "ttgo-t-beam-v1_2".)

Great software! Keep on! 

73, Werner HB9TLY

closed 2024-06-02 by richonguzman