richonguzman/LoRa_APRS_Tracker #112: Power down function HELTEC Wireless Tracker

PC9DB, 2024-05-23:
Hello,

The power down function doesn't seem to work on the HELTEC Wireless Tracker.
Probably some other boards have more buttons?

Software power down would be handy when installing the tracker into a case without a physical on/off switch.

Best regards,
Mark

IZ1HMT, 2024-05-31:
I also agree with this request. This feature would be useful, like in the Meshstatic software.
Best regards
Matteo IZ1HMT

richonguzman, 2024-05-31:
this was yesterday added...

all NON T-Beam boards will go into DEEP SLEEP using less than 1mA of current (as there is no physical switch you should know that eventually if the board is left un-used the battery will dischard fully and be damaged) but a less than 1mA current consumption rate it should last easy more than 80 days with 2200mA battery and still be ok)

to powerOff: triple press user button

last firmware = 2024.05.30

closed 2024-05-31 by richonguzman

PC9DB, 2024-06-01:
Thank you very much (also for pointing out the 1mA current while "off")

Best regards and keep up the good work!
Mark PC9DB