richonguzman/LoRa_APRS_Tracker #116: No power ON under batery t-beam 1.2 AXP2101

dcardenosa, 2024-05-30:
Hi
I have 2 T-Beams 1.2 with AXP2101 that work ok under usb power but doesn't under battery power. They don´t power up pressing the pwr button. The button works on usb power.  With usb power they measure the battery voltake ok and show (charging) message. If I disconnect the usb cable the t-beam powers off.

With the Meshtastic  fw the batery power works fine on those boards.

I set the environment on vscode to "ttgo-t-beam-v1_2". Cleaned up the vscode environment few times and make a clean git clone of the repository.

Not sure what´s the problem. :S

I also have 2 T-Beams 1.1 that work ok on batery power. 

Thanks 
David EA1FAQ

[image]

[image]

richonguzman, 2024-05-30:
Hi

Sometimes AXP2101 power management chip needs a reboot

Take out the battery and put them back on (without usb)

This may solve this issue

dcardenosa, 2024-05-30:
Yes!!! The problem was solved! 
Thanks!

David EA1FAQ

closed 2024-05-30 by richonguzman