richonguzman/LoRa_APRS_Tracker #117: Haltec Wireless Tracker 1.1 No GPS data

Dario-smg, 2024-05-31:
Hi! I use Haltec Wireless Tracker 1.1 which, unlike 1.0, requires some changes in the IO pinout. The manufacturer informs: "Confirm your hardware version, V1.1 has changed the power control pin of the GNSS chip (GPIO 3 is set to HIGH on). Please carefully review the [schematic](https://resource.heltec.cn/download/Wireless_Tracker) or refer to the corresponding hardware version’s example program". I'm just an ordinary radio amateur and I'm not very good at programming. Please help me change the program source configuration. I'd rather not burn something by giving the wrong configuration.

richonguzman, 2024-05-31:
Hi current version of HELTEC WIRELESS TRACKER is for V1.1 

so you dont need to change a bit of the code to make it run ;)

Mane76, 2024-05-31:
Hello Dario, no need for updating code, the version of Ricardo adresses V1.1.
Give it some time, for first fix it may need more than 10mins..

Dario-smg, 2024-05-31:
Thank you for your answer. Unfortunately, more hours pass and the GPS still does not get the "fix" even though I installed an additional GPS antenna, the only thing that works is GPS time synchronization. So if there is native code support for v1.1, I look elsewhere for a solution. Thank you @Ricardo and "Others" for your contribution to the development of LoRa APRS 73!

richonguzman, 2024-05-31:
> Thank you for your answer. Unfortunately, more hours pass and the GPS still does not get the "fix" even though I installed an additional GPS antenna, the only thing that works is GPS time [...]

Dario, 

if you have Time syncronization and Date synchronization the GPS is working.

although the GPS fix takes a lot of time on the first boot as it needs to download from GPS the ALMANAC of satellites.

so leave it outside (not inside house/building) for a while to get the GPS FIX complete.

closed 2024-06-02 by Dario-smg