richonguzman/LoRa_APRS_Tracker #119: Aprsdroid TNC - half of Bluetooth frames are missing on radio

sq2mo, 2024-06-04:
I am using Aprsdroid to send the frames over Bluetooth using Lora32 v2.1 TNC. It works intermittently, as some of the frames are not appearing on the radio.
I've leveraged the built-in debug capabilities to look into the Bluetooth frame and how it is processed by software. Here is the output:

Not OK - frame not sent to LoRa:
```
[DEBUG][bluetooth] Received buffer size 2. Nmea=0. �
[DEBUG][bluetooth] [1/2] c0 -> �
[DEBUG][bluetooth] [2/2] 00 -> 
[DEBUG][Bluetooth] Status: 30
[DEBUG][bluetooth] Received buffer size 38. Nmea=0. ����
[DEBUG][bluetooth] [1/38] 82 -> �
[DEBUG][bluetooth] [2/38] a0 -> �
```

OK - frame sent to LoRa:
```
[DEBUG][bluetooth] Received buffer size 40. Nmea=0. �
[DEBUG][bluetooth] [1/40] c0 -> �
[DEBUG][bluetooth] [2/40] 00 -> 
[DEBUG][bluetooth] [3/40] 82 -> �
[DEBUG][bluetooth] [4/40] a0 -> �
[DEBUG][bluetooth] [5/40] 88 -> �
[DEBUG][bluetooth] [6/40] a4 -> �
[DEBUG][bluetooth] [7/40] 62 -> b
[DEBUG][bluetooth] [8/40] 6c -> l
```

So it seems that some of the frames are split into two separate chunks, therefore not being recognized as a KISS frame and not sent to radio.

I have this issue with two different boards - TTGO LoRa32 v2.1 and DIY ESP32-WROOM-32D.
Currently using latest main code. Checked out one of the 2023.09 commits, same issue.

richonguzman, 2024-06-04:
please test with TNC2 as configuration on the APRSDroid app

sq2mo, 2024-06-04:
Yes, TNC2 works and I am currently using it as a workaround.

richonguzman, 2024-06-04:
TNC2 is the way to use it ;)

closed 2024-06-04 by richonguzman

sq2mo, 2024-06-04:
That should work, thank you. Perhaps worth adding this to wiki.

richonguzman, 2024-06-04:
https://github.com/richonguzman/LoRa_APRS_Tracker/wiki/06.-Bluetooth-Connection-%23-Conexión-Bluetooth

its there for a while now ;)

sq2mo, 2024-06-04:
Right, but there's no info about TNC2 preferred over KISS.