richonguzman/LoRa_APRS_Tracker #121: T-Deck broken?

S57PNX, 2024-06-07:
Can anybody confirm that their T-Deck still works after last code changes? I updated an early May FW to latest source code, and it doesn't boot anymore - symptoms are constand disconnect and reconnect to usb, nothing on screen.

I had to boot it in the bootloader mode (pushing trackpad during power up), then uploaded Firmware and Filesystem. Won't start up afterwards with this rapid USB disconnect and reconnect. I even installed meshtastic just to double check the hardware and it works OK.

richonguzman, 2024-06-24:
hi, 

did you by chance test it a little further?

S57PNX, 2024-06-24:
I rebuilt from the latest code and tried again, same behaviour. The keyboard backlight blinks shortly and nothing comes on screen, I can't use Monitor because the USB port quickly disconnects and reconnects (less than 1 second) continuiusly. I get the windows USB connect/disconnect sound uninterrupted.

richonguzman, 2024-06-24:
did you use the "enable" flashing process with the buttons:

use/boot button (middle button of the joystick) and RESET?

as recomended for any ESP32S3 board?

S57PNX, 2024-06-25:
Yes, I put the t-deck in the flashing mode by holding joystick button while resetting. The board connects in a stable mode and I am able to do a full upload of firmware and filesystem, no error messages. After I reset it, it goes in normal operation mode, but it doesn't boot and does rapid cycle of usb disconnect/reconnect. 

This unit worked OK with a firmware from early May.

Can anybody else test with their t-deck?

richonguzman, 2024-06-27:
I've tested and it was broken with Espressif version 6.7.0

so after a few mods and T-deck to Espressif version 6.3.1 it works again

closed 2024-06-27 by richonguzman

S57PNX, 2024-06-27:
Confirm, this fixed it for me too! Thank you.