richonguzman/LoRa_APRS_Tracker #123: Problema Heltec Tracker Lora

droptic, 2024-06-10:
Al cambiar entre las distintas configuraciónes, el icono que se ve en la pantalla no se refresca y se sobre escribe con el anterior no pudiendo saber claramente cuál configuración está en uso.
Saludos

richonguzman, 2024-06-11:
si, es una pifia de como funcionan los TFT y la velocidad de refresco, 

no me he metido mas a ellos buscando meterle mejoras a otras cosas :/

ON7MHZ, 2024-06-14:
Hi , 
might be a stupid question (and maybe not the correct place to ask )  ... but what type of board should I select for the Heltec Tracker at "import arduino project" ? 
I followed the steps as indicated , but uploading the file system failed on my Heltec Tracker with " error code 1 " ... 
Any suggestions ? 
thanks 
Pieter  ( newbie at LoRa APRS ) 
ON7MHZ

ON2ON-HAM, 2024-06-15:
Pieter,

Als ik je bericht lees 'import arduino project', is mijn eerste vraag, zit je in Visual Code te werken ? Want met die software gaan we de code Builden en Uploaden naar je bordje.

En 2e punt, ging de 'Build' voor je Heltec Wireless Tracker in General en Filesystem goed ?

Grtjs
Luc ON2ON

closed 2024-10-01 by richonguzman