richonguzman/LoRa_APRS_Tracker #124: Need Instructions to compile and download to device

wb2osz, 2024-06-11:
I followed the instructions as far as configuring the device.
The next link goes to devices supported.
How do we compile and download into the device?
Thanks.

richonguzman, 2024-06-11:
hi

did you follow this [tutorial](https://github.com/richonguzman/LoRa_APRS_Tracker/wiki/01.-Installation-Guide-%23-Guia-de-Instalacion) ?

wb2osz, 2024-06-11:
Yes. I got to the end of this page for the configuration.
https://github.com/richonguzman/LoRa_APRS_Tracker/wiki/02.-Tracker-Configuration--%23--Configuracion-del-Tracker

The bottom of the page has  [Next Boards Supported / Siguiente Placas Soportadas](https://github.com/richonguzman/LoRa_APRS_Tracker/wiki/03.-Supported-Boards-and-Environment-Selection-%23-Placas-soportadas-y-seleccion-del-entorno)

This goes back to the home page.
Instructions to compile and download to LoRa device seem to be missing.

Thanks.

wb2osz, 2024-06-11:
Oh, I see.  On the right edge is a list of pages. 
I need page 3.
So the issue is that the link at the bottom of page 2 should link to page 3.

richonguzman, 2024-06-11:
this is the next step (I edited the webpage link to continue the process)

https://github.com/richonguzman/LoRa_APRS_Tracker/wiki/03.-Upload-Firmware-and-Filesystem-%23-Subir-Firmware-y-sistema-de-archivos

closed 2024-06-11 by wb2osz