richonguzman/LoRa_APRS_Tracker #125: reboot loop

dg1kwa, 2024-06-14:
What happen?

--- Terminal on COM6 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (Sv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
[INFO][AXP192] init done!
[INFO][Main] RichonGuzman (CA2RXU) --> LoRa APRS Tracker/Station
[INFO][Main] Version: 2024.06.04
[INFO][LoRa] Initializing SX127X ...
[INFO][LoRa] LoRa init done!
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1). 

Core  1 register dump:
PC      : 0x4008672a  PS      : 0x00050035  A0      : 0x4023ba9e  A1      : 0x3ffbf82c
A2      : 0x00000000  A3      : 0x00000009  A4      : 0x00000016  A5      : 0x3ffbf7bc
A6      : 0x00000000  A7      : 0x3ffbdf04  A8      : 0x00000000  A9      : 0x00000000
A10     : 0x3ffbdf00  A11     : 0x00000000  A12     : 0x00000001  A13     : 0x3ffbdf00
A14     : 0x00000003  A15     : 0x00060023  SAR     : 0x00000000  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000
Core  1 was running in ISR context:
EPC1    : 0x401252eb  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x00000000

Backtrace: 0x40086727:0x3ffbf82c |<-CORRUPTED

Core  0 register dump:
PC      : 0x4023ba9e  PS      : 0x00060f35  A0      : 0x80123d74  A1      : 0x3ffbb1d0
A2      : 0x00000000  A3      : 0x40092940  A4      : 0x00060820  A5      : 0x3ffcf4d0
A6      : 0x007bf848  A7      : 0x003fffff  A8      : 0x8012384e  A9      : 0x3ffbb1a0  
A10     : 0x00000000  A11     : 0x80000001  A12     : 0x8009978c  A13     : 0x3ffbb0d0
A14     : 0x00000003  A15     : 0x00060023  SAR     : 0x0000001a  EXCCAUSE: 0x00000006
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000

Backtrace: 0x4023ba9b:0x3ffbb1d0 0x40123d71:0x3ffbb1f0 0x400971d4:0x3ffbb210
  #0  0x4023ba9b in cpu_ll_waiti at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/hal/esp32/include/hal/cpu_ll.h:183
      (inlined by) esp_pm_impl_waiti at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_pm/pm_impl.c:853
  #1  0x40123d71 in esp_vApplicationIdleHook at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/freertos_hooks.c:63
  #2  0x400971d4 in prvIdleTask at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c:4099

richonguzman, 2024-06-14:
Hi

I need lots of info:
Which board?
Which env selected?

Firmware uploaded?
Tracker_conf.json edited?
Filesystem uploaded?

dg1kwa, 2024-06-15:
Hi,
it is a LILYGO T-Beam 433MHz (picture at this email).

I try with ttgo-t-beam-v1_2 and ttgo-t-beam-v1

Firmware yes, Tracker_conf.json yes and filesystem yes.

Danke und viele Grüße
  Andreas Wißkirchen

---
Andreas Wißkirchen DG1KWA/HS0ZNJ

Achtung!
Neue Anschrift:
Hans-Scholl-Str.1
40789 Monheim am Rhein / Germany

> Hi
> 
> I need lots of info:
> Which board?
> Which env selected?
> 
> Firmware uploaded?
> Tracker_conf.json edited?
> Filesystem uploaded?
> 
> [2].

Links:
------
[1] 
https://github.com/richonguzman/LoRa_APRS_Tracker/issues/125#issuecomment-2168844329
[2] 
https://github.com/
--=_2c4c254dde6812549b33e584d2b50fb2
Content-Type: multipart/related;
 boundary="=_c40d2aa8fccdadc114d656932b8dba6b"

--=_c40d2aa8fccdadc114d656932b8dba6b
Content-Transfer-Encoding: quoted-printable
Content-Type: text/html; charset=UTF-8

<html><head><meta http-equiv=3D"Content-Type" content=3D"text/html; charset=
=3DUTF-8" /></head><body style=3D'font-size: 10pt; font-family: Verdana,Gen=
eva,sans-serif'>
<p>Hi,<br />it is a <span id=3D"productTitle" class=3D"a-size-large pr=
oduct-title-word-break">LILYGO T-Beam 433MHz (picture at this email).<br />=
</span></p>
<p>I try with ttgo-t-beam-v1_2 and ttgo-t-beam-v1</p>
<p>Firmware yes, Tracker_conf.json yes and filesystem yes.</p>
<div id=3D"signature">
<div class=3D"pre" style=3D"margin: 0; padding: 0; font-family: monospace">=
Danke und viele Gr&uuml;&szlig;e<br /> Andreas Wi&szlig;kirchen<br /><=
br /><br />---<br />Andreas Wi&szlig;kirchen DG1KWA/HS0ZNJ<br /><br />Achtu=
ng!<br />Neue Anschrift:<br />Hans-Scholl-Str.1<br />40789 Monheim am Rhein=
 / Germany</div>
</div>
<p><br /></p>
<p id=3D"reply-intro">Am 2024-06-15 00:34, schrieb Ricardo Guzman (Richongu=
zman):</p>
<blockquote type=3D"cite" style=3D"padding: 0 0.4em; border-left: #1010ff 2=
px solid; margin: 0">
<div id=3D"replybody1">
<p><br /></p>
<p dir=3D"auto">Hi</p>
<p dir=3D"auto">I need lots of info:<br />Which board?<br />Which env selec=
ted?</p>
<p dir=3D"auto">Firmware uploaded?<br />Tracker_conf.json edited?<br />File=
[... 811 lines cut ...]
0y1TRUMAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAKZGVz
YwAAAPwAAABkY3BydAAAAWAAAAB8d3RwdAAAAdwAAAAUY2hhZAAAAfAAAAAsclhZWgAAAhwAAAAU
Z1hZWgAAAjAAAAAUYlhZWgAAAkQAAAAUclRSQwAAAlgAAAAgZ1RSQwAAAlgAAAAgYlRSQwAAAlgA
AAAgbWx1YwAAAAAAAAABAAAADGVuVVMAAABGAAAAHABEAEMASQAtAFAAMwAgAEQANgA1ACAARwBh
AG0AdQB0ACAAdwBpAHQAaAAgAHMAUgBHAEIAIABUAHIAYQBuAHMAZgBlAHIAAG1sdWMAAAAAAAAA
AQAAAAxlblVTAAAAYAAAABwAQwBvAHAAeQByAGkAZwBoAHQAIAAoAGMAKQAgADIAMAAyADIAIABT
AGEAbQBzAHUAbgBnACAARQBsAGUAYwB0AHIAbwBuAGkAYwBzACAAQwBvAC4ALAAgAEwAdABkAC5Y
WVogAAAAAAAA9tYAAQAAAADTLXNmMzIAAAAAAAEMPQAABdz///MrAAAHkAAA/ZD///uj///9owAA
A9oAAMCMWFlaIAAAAAAAAIPcAAA9vf///7tYWVogAAAAAAAASr0AALE2AAAKuVhZWiAAAAAAAAAo
PQAAEQwAAMjYcGFyYQAAAAAAAwAAAAJmZgAA8qcAAA1ZAAAT0AAAClv/5Bl6AAAAAAAAAACurq6u
sgEAALMEAAAVCgAAeAMAAHb+//8AAAAAAQAbAAAAAABDCAAAeQgAAAYAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAADAAAAAAAAAAAZ
AAAAGQAAAAAAAAAAAAAAAAAAAAAAAJQAAABkAwAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAARwAAABAAAAAuAAAAQggAAB0AAABmCAAA
4AEAAAAAAAAAAAAAAAAAACAAAAAgAwAAGgAAAAAAAABR3K8BAAAAAHb+//+zBAAA7u7u7mEBAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAADEAAAAAAAAAAAAAAAAAAAAAAAAAXAEAAAAA
AAAAAAAAMwEAAAEAAAAAAAAAAAAAAAAAAAAAAAAAfQAAAF6WAAAQAAAAEAAAAF6WAAAQAAAAAAAA
AAAAAAB6CgAATAgAAHoKAAAAAAAAAAAAAPr///90AAAA9QAAAC0AAACfCAAAdwgAAI4HAABCCAAA
vAgAAKcIAACFCAAAYwgAAOsIAAAUCAAAMQgAADkIAACCCAAAUQgAAD4IAAAeCAAA1AgAAOoHAAAR
CAAAiAcAAIUIAADGBwAAhAcAAHYHAADGBwAA1AcAAIQHAAAsCAAAugcAAA4IAACWCAAAwQYAAKUG
AAC+BgAA/QYAAHUIAACNCQAA/wgAAJsIAADgBwAAvgYAAMgGAADrBgAAOwcAADQIAABVCQAAzwcA
AJIHAAClBwAA+gYAAE4HAAA2CAAAEQgAABIIAAAUCAAAGQgAADMIAACVCAAAeQgAADsHAAB0CAAA
fQgAAH4IAACHCAAAhggAANwGAACZCAAAmwgAANAHAAC2CAAAigAAAAAAAAACAAAAACYAAOr///8A
AAAAAAAAAAAAAAAZAAAAHQAAAAAAAAAAAAAAAAAAAAIAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAE
AAAACAAA

richonguzman, 2024-06-15:
sorry the Picture didn't load as needed

but you can know the "real" board to use by reading the text near the SMA antenna.

its V1.1 or V1.0 or V1.2 but all of them need to be selected specifically or they will fail

PS: usually the LoRa fail error in this case is because it wasn't loaded the filesystem too

dg1kwa, 2024-06-16:
ohh ... sorry

It is a T22_V1.1 (20191212)

Danke und viele Grüße
  Andreas Wißkirchen

---
Andreas Wißkirchen DG1KWA/HS0ZNJ

Achtung!
Neue Anschrift:
Hans-Scholl-Str.1
40789 Monheim am Rhein / Germany

> sorry the Picture didn't load as needed
> 
> but you can know the "real" board to use by reading the text near the 
> SMA antenna.
> 
> its V1.1 or V1.0 or V1.2 but all of them need to be selected 
> specifically or they will fail
> 
> PS: usually the LoRa fail error in this case is because it wasn't 
> loaded the filesystem too
> 
> [2].

Links:
------
[1] 
https://github.com/richonguzman/LoRa_APRS_Tracker/issues/125#issuecomment-2169439370
[2] 
https://github.com/
--=_6378ca14fcc3048110ac806daeb43657
Content-Type: multipart/related;
 boundary="=_8f3b647189ea7b9d5dbc77aeffc9d9ae"

--=_8f3b647189ea7b9d5dbc77aeffc9d9ae
Content-Transfer-Encoding: quoted-printable
Content-Type: text/html; charset=UTF-8

<html><head><meta http-equiv=3D"Content-Type" content=3D"text/html; charset=
=3DUTF-8" /></head><body style=3D'font-size: 10pt; font-family: Verdana,Gen=
eva,sans-serif'>
<p>ohh ... sorry</p>
<p>It is a T22_V1.1 (20191212)</p>
<div id=3D"signature">
<div class=3D"pre" style=3D"margin: 0; padding: 0; font-family: monospace">=
Danke und viele Gr&uuml;&szlig;e<br /> Andreas Wi&szlig;kirchen<br /><=
br /><br />---<br />Andreas Wi&szlig;kirchen DG1KWA/HS0ZNJ<br /><br />Achtu=
ng!<br />Neue Anschrift:<br />Hans-Scholl-Str.1<br />40789 Monheim am Rhein=
 / Germany</div>
</div>
<p><br /></p>
<p id=3D"reply-intro">Am 2024-06-15 14:19, schrieb Ricardo Guzman (Richongu=
zman):</p>
<blockquote type=3D"cite" style=3D"padding: 0 0.4em; border-left: #1010ff 2=
px solid; margin: 0">
<div id=3D"replybody1">
<p><br /></p>
<p dir=3D"auto">sorry the Picture didn't load as needed</p>
<p dir=3D"auto">but you can know the "real" board to use by reading the tex=
t near the SMA antenna.</p>
<p dir=3D"auto">its V1.1 or V1.0 or V1.2 but all of them need to be selecte=
d specifically or they will fail</p>
<p dir=3D"auto">PS: usually the LoRa fail error in this case is because it =
wasn't loaded the filesystem too</p>
<p style=3D"font-size: small; -webkit-text-size-adjust: none; color: #666;"=
ichonguzman/LoRa_APRS_Tracker/issues/125#issuecomment-2169439370" target=3D=
"_blank" rel=3D"noopener noreferrer">view it on GitHub</a>, or <a href=3D"h=
WNNAVCNFSM6AAAAABJK6XXH6VHI2DSMVQWIX3LMV43OSLTON2WKQ3PNVWWK3TUHMZDCNRZGQZTS=
MZXGA" target=3D"_blank" rel=3D"noopener noreferrer">unsubscribe</a>.<br />=
yle=3D"color: transparent; font-size: 0; display: none; visibility: hidden;=
 overflow: hidden; opacity: 0; width: 0px; height: 0px; max-width: 0; max-h=
ker/issues/125/2169439370</span><span>@</span><span>github</span><span>.</s=
pan><span>com&gt;</span></span></p>
</div>
</blockquote>
</body></html>

--=_8f3b647189ea7b9d5dbc77aeffc9d9ae
Content-Transfer-Encoding: base64
Content-ID: ***@***.***>
Content-Type: image/gif;
 name=blocked.gif
Content-Disposition: inline;
 filename=blocked.gif;
 size=118

R0lGODlhZAAyAIAAAPrOzgAAACH5BAAAAAAALAAAAABkADIAAAJNhI+py+0Po5y02ouz3rz7D4bi
SJbmiabqyrbuC8fyTNf2jef6zvf+DwwKh8Si8YhMKpfMpvMJjUqn1Kr1is1qt9yu9wsOi8fksvls
KwAAOw==
--=_8f3b647189ea7b9d5dbc77aeffc9d9ae--

--=_6378ca14fcc3048110ac806daeb43657--

OK2MOP, 2024-06-17:
Double check you are building for ttgo-t-beam-v1 (if you do not have also SX126x chip on board) not the default one, (which is ttgo-t-beam-v1_2), and that the configuration file is correct and you have uploaded it. 

I was fighting with boot loops using older version of config file for two days, compiling different versions of firmware (add to that the present issue of aprs.fi with the normal compressed packets - mic-E packets work when I returned back to old firmware and settings) and in the end I just copied all the settings to new configuration file to make it work. 

In my case the probable culprit was the lora section, in configuration file, which was originally an object with values in JSON, but in present version is an array of objects.

richonguzman, 2024-06-17:
> Double check you are building for ttgo-t-beam-v1 (if you do not have also SX126x chip on board) not the default one, (which is ttgo-t-beam-v1_2), and that the configuration file is correct and you [...]

the way its saved on the json file does not change the way it works

from the start to use Tracker Firmware there is a three process way to make it work:

1)select the environment for the board you will use
2)upload firmware (the program the tracker runs)
3)upload filetsystem (the json you edited with your calling and preferences)

dg1kwa, 2024-06-18:
ok, tnx. I try later again but I sure that I use same way.

73 de DG1KWA / Andreas

Danke und viele Grüße
  Andreas Wißkirchen

---
Andreas Wißkirchen DG1KWA/HS0ZNJ

Achtung!
Neue Anschrift:
Hans-Scholl-Str.1
40789 Monheim am Rhein / Germany

>> Double check you are building for ttgo-t-beam-v1 (if you do not have 
>> also SX126x chip on board) not the default one, (which is 
>> ttgo-t-beam-v1_2), and that the configuration file is correct and you 
>> have uploaded it.
>> 
>> I was fighting with boot loops using older version of config file for 
>> two days, compiling different versions of firmware (add to that the 
>> present issue of aprs.fi with the normal compressed packets - mic-E 
>> packets work - when I returned back to old firmware and settings) and 
>> in the end I just copied all the settings to new configuration file to 
>> make it work.
>> 
>> In my case the probable culprit was the lora section, in configuration 
>> file, which was originally an object with values in JSON, but in 
>> present version is an array of objects.
> 
> the way its saved on the json file does not change the way it works
> 
> from the start to use Tracker Firmware there is a three process way to 
> make it work:
> 
> 1)select the environment for the board you will use
> 2)upload firmware (the program the tracker runs)
> 3)upload filetsystem (the json you edited with your calling and 
> preferences)
> 
> [2].

Links:
------
[1] 
https://github.com/richonguzman/LoRa_APRS_Tracker/issues/125#issuecomment-2173894007
[2] 
https://github.com/
--=_bff0b1b33299bec28d443fcbb069a833
Content-Type: multipart/related;
 boundary="=_12572a5e19ba4d9eded227497b64d4d0"

--=_12572a5e19ba4d9eded227497b64d4d0
Content-Transfer-Encoding: quoted-printable
Content-Type: text/html; charset=UTF-8

<html><head><meta http-equiv=3D"Content-Type" content=3D"text/html; charset=
=3DUTF-8" /></head><body style=3D'font-size: 10pt; font-family: Verdana,Gen=
eva,sans-serif'>
<p>ok, tnx. I try later again but I sure that I use same way.</p>
<p><br /></p>
<p>73 de DG1KWA / Andreas</p>
<div id=3D"signature">
<div class=3D"pre" style=3D"margin: 0; padding: 0; font-family: monospace">=
Danke und viele Gr&uuml;&szlig;e<br /> Andreas Wi&szlig;kirchen<br /><=
br /><br />---<br />Andreas Wi&szlig;kirchen DG1KWA/HS0ZNJ<br /><br />Achtu=
ng!<br />Neue Anschrift:<br />Hans-Scholl-Str.1<br />40789 Monheim am Rhein=
 / Germany</div>
</div>
<p><br /></p>
<p id=3D"reply-intro">Am 2024-06-17 18:55, schrieb Ricardo Guzman (Richongu=
zman):</p>
<blockquote type=3D"cite" style=3D"padding: 0 0.4em; border-left: #1010ff 2=
px solid; margin: 0">
<div id=3D"replybody1">
<p><br /></p>
<blockquote type=3D"cite" style=3D"padding: 0 0.4em; border-left: #1010ff 2=
px solid; margin: 0">
<p dir=3D"auto">Double check you are building for ttgo-t-beam-v1 (if you do=
 not have also SX126x chip on board) not the default one, (which is ttgo-t-=
beam-v1_2), and that the configuration file is correct and you have uploade=
d it.</p>
<p dir=3D"auto">I was fighting with boot loops using older version of confi=
g file for two days, compiling different versions of firmware (add to that =
the present issue of aprs.fi with the normal compressed packets - mic-E pac=
kets work - when I returned back to old firmware and settings) and in the e=
nd I just copied all the settings to new configuration file to make it work=
=2E</p>
<p dir=3D"auto">In my case the probable culprit was the lora section, in co=
nfiguration file, which was originally an object with values in JSON, but i=
n present version is an array of objects.</p>
</blockquote>
<p dir=3D"auto">the way its saved on the json file does not change the way =
it works</p>
<p dir=3D"auto">from the start to use Tracker Firmware there is a three pro=
cess way to make it work:</p>
<p dir=3D"auto">1)select the environment for the board you will use<br />2)=
upload firmware (the program the tracker runs)<br />3)upload filetsystem (t=
he json you edited with your calling and preferences)</p>
<p style=3D"font-size: small; -webkit-text-size-adjust: none; color: #666;"=
ichonguzman/LoRa_APRS_Tracker/issues/125#issuecomment-2173894007" target=3D=
"_blank" rel=3D"noopener noreferrer">view it on GitHub</a>, or <a href=3D"h=
IHNAVCNFSM6AAAAABJK6XXH6VHI2DSMVQWIX3LMV43OSLTON2WKQ3PNVWWK3TUHMZDCNZTHA4TI=
MBQG4" target=3D"_blank" rel=3D"noopener noreferrer">unsubscribe</a>.<br />=
yle=3D"color: transparent; font-size: 0; display: none; visibility: hidden;=
 overflow: hidden; opacity: 0; width: 0px; height: 0px; max-width: 0; max-h=
ker/issues/125/2173894007</span><span>@</span><span>github</span><span>.</s=
pan><span>com&gt;</span></span></p>
</div>
</blockquote>
</body></html>

--=_12572a5e19ba4d9eded227497b64d4d0
Content-Transfer-Encoding: base64
Content-ID: ***@***.***>
Content-Type: image/gif;
 name=blocked.gif
Content-Disposition: inline;
 filename=blocked.gif;
 size=118

R0lGODlhZAAyAIAAAPrOzgAAACH5BAAAAAAALAAAAABkADIAAAJNhI+py+0Po5y02ouz3rz7D4bi
SJbmiabqyrbuC8fyTNf2jef6zvf+DwwKh8Si8YhMKpfMpvMJjUqn1Kr1is1qt9yu9wsOi8fksvls
KwAAOw==
--=_12572a5e19ba4d9eded227497b64d4d0--

--=_bff0b1b33299bec28d443fcbb069a833--

RadioMan57, 2024-06-22:
Reboot loop also here, due to a home-made syntax error in tracker_config.json.
There was an extra comma in the bme section. VSCode flagged it.
Error corrected, problem gone.
73, Markus HB9BRJ

richonguzman, 2024-06-25:
has @dg1kwa solved his issue?

dg1kwa, 2024-06-25:
> has @dg1kwa [1] solved his issue?

I try next week because I'm not at home this week.

vy 73 de DG1KWA/HS0ZNJ Andreas

Links:
------
[1] https://github.com/dg1kwa
[2] 
https://github.com/richonguzman/LoRa_APRS_Tracker/issues/125#issuecomment-2189001460
[3] 
https://github.com/

steffenspies, 2024-08-27:
@dg1kwa do you use  a backslash "\" in your tracker_config.json file for the symbol you choose?
I realized that this leads to a reboot loop.

`			"symbol": "[",
			"overlay": "/",
`
If I set "overlay" to "\" it will reboot.

73 de Steffen, DL1SX

closed 2024-10-01 by richonguzman