richonguzman/LoRa_APRS_Tracker #130: mismatch between json config file and displayed data at aprs.fi

dg1rue, 2024-06-22:
Hey Richon, thank you so much for your effots in programmin LoRa APRS.
I use TTGO  T22_V1.1 which is recognized by VSC as 1.2
I attach the configuration json file and reception screenshots of aprs.fi
As you can see there is a little difference between config and reception.
May be this information helps you to improve your software.

Hope to hear from you.
best regards and 73s de DG1RUE (aka Holger)
[image]
[image]
[image]
[tracker_config.json](https://github.com/user-attachments/files/15937485/tracker_config.json)

richonguzman, 2024-06-22:
All looks good in configuration 

It seems DG1RUE-15 may be joining two packets and this is appearing as faulty comment

Can you update it to my last igate firmware?

richonguzman, 2024-06-25:
could we start all over again?

do you have updated the igate?

closed 2024-10-01 by richonguzman