richonguzman/LoRa_APRS_Tracker #131: How to Power Off (new Heltec LoRa Tracker PCB)?

g3cwi, 2024-06-23:
I have the newer Heltech LoRa Tracker board. This has just a P and R button. I can't figure out how to power down the tracker. It's in a case with a battery so disconnecting the power is not really a good option. I did try to use the Power Off option in the menu. This is tricky to reach and also does not allow me to power the board off. Ideas?

richonguzman, 2024-06-23:
As this board has no switch or power management chip: pressing 3 times USER button will make the board go into deep sleep for over a month

This is the only available way to make it use very little current (0,7mA/H)

To power off for real, use an external switch

g3cwi, 2024-06-23:
Perfect thanks. I noted that the reset switch will wake it up again. Thank you for your work on these devices.

closed 2024-06-24 by richonguzman