richonguzman/LoRa_APRS_Tracker #132: Support for TTGO T-Beam v1.2 with SX1276 (915 MHz)

wb2osz, 2024-06-30:
Kindly consider adding support for TTGO T-Beam v1.2 with SX1276 for 915 MHz.
I was able to get it working with the following diffs.
platformio.ini had sections for TTGO_T_Beam_V1_2 and TTGO_T_Beam_V1_2_SX1262.
I added a similar section for TTGO_T_Beam_V1_2_SX1276.
Your instructions did not mention changing the "[platformio] default_envs" line but I figured that one out.

```
diff --git a/platformio.ini b/platformio.ini
index fcc9032..e7cf5b6 100644
--- a/platformio.ini
+++ b/platformio.ini
@@ -9,7 +9,7 @@
 ; https://docs.platformio.org/page/projectconf.html
 
 [platformio]
-default_envs = ttgo-t-beam-v1_2
+default_envs = ttgo-t-beam-v1_2_SX1276
 
 [env]
 framework = arduino
@@ -60,6 +60,23 @@ lib_deps =
 	adafruit/Adafruit SH110X @ 2.1.10
 
 
+[env:ttgo-t-beam-v1_2_SX1276]
+extends = env:esp32
+board = ttgo-t-beam
+build_flags =
+	-Werror -Wall
+	-DTTGO_T_Beam_V1_2_SX1276
+	-DHAS_SX1276
+	-DHAS_AXP2101
+	-DHAS_BT_CLASSIC
+lib_deps =
+	${common.lib_deps}
+	lewisxhe/XPowersLib @ 0.2.4
+	adafruit/Adafruit GFX Library @ 1.11.9
+	adafruit/Adafruit SSD1306 @ 2.5.10
+	adafruit/Adafruit SH110X @ 2.1.10
+
+
 [env:ttgo-t-beam-v1_2_SX1262]
 extends = env:esp32
 board = ttgo-t-beam
diff --git a/src/boards_pinout.h b/src/boards_pinout.h
index 6dfd393..67e010f 100644
--- a/src/boards_pinout.h
+++ b/src/boards_pinout.h
@@ -6,7 +6,7 @@
 #undef OLED_RST
 
 
-#if defined(TTGO_T_Beam_V1_2) || defined(TTGO_T_Beam_V1_0)
+#if defined(TTGO_T_Beam_V1_2) || defined(TTGO_T_Beam_V1_0) || defined(TTGO_T_Beam_V1_2_SX1276)
     #define OLED_SDA            21
     #define OLED_SCL            22
     #define OLED_RST            16
diff --git a/src/menu_utils.cpp b/src/menu_utils.cpp
index c7b3de1..bbe2153 100644
--- a/src/menu_utils.cpp
+++ b/src/menu_utils.cpp
@@ -620,7 +620,7 @@ namespace MENU_Utils {
                             sixthRowMainMenu += "mA";
                         }
                     #endif
-                    #if defined(TTGO_T_Beam_V1_2) || defined(TTGO_T_Beam_V1_2_SX1262) || defined(TTGO_T_Beam_S3_SUPREME_V3)
+                    #if defined(TTGO_T_Beam_V1_2) || defined(TTGO_T_Beam_V1_2_SX1262) || defined(TTGO_T_Beam_V1_2_SX1276) || defined(TTGO_T_Beam_S3_SUPREME_V3)
                         if (Config.notification.lowBatteryBeep && !POWER_Utils::isCharging() && batteryCharge.toInt() < lowBatteryPercent) {
                             lowBatteryPercent = batteryCharge.toInt();
                             NOTIFICATION_Utils::lowBatteryBeep();
diff --git a/src/power_utils.cpp b/src/power_utils.cpp
index 700522c..e8c9f10 100644
--- a/src/power_utils.cpp
+++ b/src/power_utils.cpp
@@ -196,7 +196,7 @@ namespace POWER_Utils {
             PMU.setLDO3Voltage(3300);
             PMU.enableLDO3();
         #endif
-        #if defined(TTGO_T_Beam_V1_2) || defined(TTGO_T_Beam_V1_2_SX1262)
+        #if defined(TTGO_T_Beam_V1_2) || defined(TTGO_T_Beam_V1_2_SX1262) || defined(TTGO_T_Beam_V1_2_SX1276)
             PMU.setALDO3Voltage(3300);
             PMU.enableALDO3(); 
         #endif
@@ -210,7 +210,7 @@ namespace POWER_Utils {
         #ifdef HAS_AXP192
             PMU.disableLDO3();
         #endif
-        #if defined(TTGO_T_Beam_V1_2) || defined(TTGO_T_Beam_V1_2_SX1262)
+        #if defined(TTGO_T_Beam_V1_2) || defined(TTGO_T_Beam_V1_2_SX1262) || defined(TTGO_T_Beam_V1_2_SX1276)
             PMU.disableALDO3();
         #endif
         #if defined(TTGO_T_Beam_S3_SUPREME_V3)
@@ -223,7 +223,7 @@ namespace POWER_Utils {
             PMU.setLDO2Voltage(3300);
             PMU.enableLDO2();
         #endif
-        #if defined(TTGO_T_Beam_V1_2) || defined(TTGO_T_Beam_V1_2_SX1262)
+        #if defined(TTGO_T_Beam_V1_2) || defined(TTGO_T_Beam_V1_2_SX1262) || defined(TTGO_T_Beam_V1_2_SX1276)
             PMU.setALDO2Voltage(3300);
             PMU.enableALDO2();
         #endif
@@ -237,7 +237,7 @@ namespace POWER_Utils {
         #ifdef HAS_AXP192
             PMU.disableLDO2();
         #endif
-        #if defined(TTGO_T_Beam_V1_2) || defined(TTGO_T_Beam_V1_2_SX1262)
+        #if defined(TTGO_T_Beam_V1_2) || defined(TTGO_T_Beam_V1_2_SX1262) || defined(TTGO_T_Beam_V1_2_SX1276)
             PMU.disableALDO2();
         #endif
         #if defined(TTGO_T_Beam_S3_SUPREME_V3)
@@ -303,7 +303,7 @@ namespace POWER_Utils {
             return result;
         #endif
 
-        #if defined(TTGO_T_Beam_V1_2) || defined(TTGO_T_Beam_V1_2_SX1262)
+        #if defined(TTGO_T_Beam_V1_2) || defined(TTGO_T_Beam_V1_2_SX1262) || defined(TTGO_T_Beam_V1_2_SX1276)
             bool result = PMU.begin(Wire, AXP2101_SLAVE_ADDRESS, I2C_SDA, I2C_SCL);
             if (result) {
                 PMU.disableDC2();
@@ -368,7 +368,7 @@ namespace POWER_Utils {
             PMU.setSysPowerDownVoltage(2600);
         #endif
 
-        #if defined(TTGO_T_Beam_V1_2) || defined(TTGO_T_Beam_V1_2_SX1262)
+        #if defined(TTGO_T_Beam_V1_2) || defined(TTGO_T_Beam_V1_2_SX1262)  || defined(TTGO_T_Beam_V1_2_SX1276)
             Wire.begin(SDA, SCL);
             if (begin(Wire)) {
                 logger.log(logging::LoggerLevel::LOGGER_LEVEL_INFO, "AXP2101", "init done!");
diff --git a/src/station_utils.cpp b/src/station_utils.cpp
index 47273a9..2f23e5d 100644
--- a/src/station_utils.cpp
+++ b/src/station_utils.cpp
@@ -438,7 +438,7 @@ namespace STATION_Utils {
                 comment += batteryChargeCurrent;
                 comment += "mA)";
             #endif
-            #if defined(TTGO_T_Beam_V1_2) || defined(TTGO_T_Beam_V1_2_SX1262) || defined(TTGO_T_Beam_S3_SUPREME_V3)
+            #if defined(TTGO_T_Beam_V1_2) || defined(TTGO_T_Beam_V1_2_SX1262) || defined(TTGO_T_Beam_V1_2_SX1276) || defined(TTGO_T_Beam_S3_SUPREME_V3)
                 comment += " Bat=";
                 comment += String(batteryVoltage.toFloat()/1000,2);
                 comment += "V (";

```

Thank you for all the work you put into this.
73,
John WB2OSZ

richonguzman, 2024-06-30:
Sure John, I 'm having this week as I was finishing some stuff over igate firmware before

richonguzman, 2024-07-01:
please test now as versions for 915 has been added for all boards

wb2osz, 2024-07-01:
Muchas gracias!

richonguzman, 2024-07-01:
> Muchas gracias!

please test and then write to know it worked for you ;)

GFaza16550, 2024-08-18:
Hola Ricardo! Felicitaciones por tu trabajo!
En caso que quisiera usar este firmware pero con la banda de 915 MHz, cual sería la configuración de frecuencias?
Gracias!

richonguzman, 2024-08-18:
Hola

Solo debes cargar la frecuencia que usen allá . La idea es que el igate y todos los trackers que estén ahí cerca usen siempre la misma

GFaza16550, 2024-08-18:
> Hola
> 
> Solo debes cargar la frecuencia que usen allá . La idea es que el igate y todos los trackers que estén ahí cerca usen siempre la misma

Gracias!

GFaza16550, 2024-08-21:
@richonguzman , logré compilar el paquete para mi Heltec V3 WiFi, pero cuando intenta subirlo me dice:

Auto-detected: COM9
Uploading .pio\build\ttgo-t-beam-v1_2\firmware.bin
esptool.py v4.5.1
Serial port COM9
Connecting....

A fatal error occurred: This chip is ESP32-S3 not ESP32. Wrong --chip argument?
*** [upload] Error 2

Que sugieres? Esta es una placa estandar.

Que puedo estar haciendo mal?

Gracias.

richonguzman, 2024-08-22:
> @richonguzman , logré compilar el paquete para mi Heltec V3 WiFi, pero cuando intenta subirlo me dice:
>
> Auto-detected: COM9 Uploading .pio\build\ttgo-t-beam-v1_2\firmware.bin esptool.py v4.5.1 [...]

eso indica que estas subiendo el firmware compilado para otra placa (una ESP32 y no la ESP32S3 que tiene la heltec v3)

la estas subiendo desde el WebFlasher, no?

GFaza16550, 2024-08-22:
Ricardo, no, estoy haciendo el paso a paso que tu tienes escrito, y selecciono Heltec v3.
Por eso lo encuentro confuso.

[image]

Gracias!

richonguzman, 2024-08-22:
lo que explica el WIKI es que ahi agregas TTGO LORA32 OLED V1

https://github.com/richonguzman/LoRa_APRS_Tracker/wiki/01.-Installation-Guide-%23-Guia-de-Instalacion

de ahi una vez que se compilado todo, cambias solo el ENVIRONMENT a la placa que tu necesitas

GFaza16550, 2024-08-22:
Pero, donde dice TTGO LORA32 OLED V1, no debería elegir mi HELTEC? 
Acabo de ver en la página de Heltec que hay dos versiones v3, una Kit (que trae el S3) y otra que no, que solo dice LoRa, y trae el ESP32 estándar.

Entonces, con esto probé tres cosas.: 

- La que dices tu (TTGO LORA32 OLED v1
- La que dice WiFi Kit v3
- La que dice WiFi Lora v3

Además, en cada una de esas compilaciones le puse el env del Heltec V3 TNC (porque no tiene GPS).

Y si bien flashea bien, y luego sube bien el file system, nunca inicia (volví a flashear y siempre queda hunged.

Extraño, no se que más probar.

richonguzman, 2024-08-22:
Porque no sigues el tutorial? Así siempre funcionará

GFaza16550, 2024-08-22:
Eso trato de hacer, pero quizás no me expreso correctamente... Cuando selecciono la placa, al importar el proyecto.. Selecciono "TTGO LORA32 OLED v1"? Esa es la correcta? Gracias.

richonguzman, 2024-08-22:
Si

Eso es solo para importar el proyecto 

Con eso listo después en ENVIRONMENT, cambias a tu placa y se ajusta todo

Nota: si tienes una heltec con esp32s3, debes apretar user + rst para que quede disponible para subir cada vez

richonguzman, 2024-08-22:
Envíame foto de cuál placa es , no debería ser tan difícil en tu caso! Animo

GFaza16550, 2024-08-22:
Adjunto fotos, es la versión de 915 MHz.
[image]
[image]

richonguzman, 2024-08-22:
ok Heltec V3 915MHz version

can you tell me the Serial Output of vscode when it boots ? (with the plug symbol on the bottom of VSCODE you could print the boot process which should give us data of the failure)

GFaza16550, 2024-08-22:
Do you want me to test booting the board on flashing mode? Or standard mode?

In way way i connect it to the USB, it shows nothing. Only keeps the orange light turned on (fixed in orange)..

The serial monitor says:

 *  Executing task: C:\Users\033462613\.platformio\penv\Scripts\platformio.exe device monitor 

Please build project in debug configuration to get more details about an exception.
See https://docs.platformio.org/page/projectconf/build_configurations.html

--- Terminal on COM9 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H

GFaza16550, 2024-08-22:
If I press ENTER:

Disconnected (ClearCommError failed (PermissionError(13, 'Access is denied.', None, 5)))
Reconnecting to COM9 
Please build project in debug configuration to get more details about an exception.
See https://docs.platformio.org/page/projectconf/build_configurations.html

         Connected!

richonguzman, 2024-08-22:
> If I press ENTER:
>
> Disconnected (ClearCommError failed (PermissionError(13, 'Access is denied.', None, 5))) Reconnecting to COM9 Please build project in debug configuration to get more details about [...]

after "Connected" it should boot up and show many messages validating each part of it

GFaza16550, 2024-08-22:
it shows nothing after the "Connected!". Strange.

richonguzman, 2024-08-22:
> it shows nothing after the "Connected!". Strange.

press just reset and see

GFaza16550, 2024-08-22:
Shows this, nothing else.

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x9 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x4bc
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a0c
entry 0x403c98d0

richonguzman, 2024-08-22:
can you confirm the board has a successful upload of filesystem too?

its really strange that you dont get any output, you should at least get what is going on!

GFaza16550, 2024-08-22:
The first try it failed (it didn't disconect the board after uploading the firmware).,
I disconntect and connected it again, and it worked. Message successful.
Should i try to change the COM port?
It's strange.

GFaza16550, 2024-08-22:
Uploading firmware:

Writing at 0x0012b3bb... (97 %)
Writing at 0x00130ae9... (100 %)
Wrote 1197840 bytes (763842 compressed) at 0x00010000 in 17.6 seconds (effective 543.1 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
====================================================== [SUCCESS] Took 44.20 seconds ======================================================

Environment                 Status    Duration
--------------------------  --------  ------------
heltec_wifi_lora_32_v3_TNC  SUCCESS   00:00:44.197
====================================================== 1 succeeded in 00:00:44.197 ======================================================

GFaza16550, 2024-08-22:
Uploading the file system:

Serial port COM12
Connecting....
Chip is ESP32-S3 (revision v0.2)
Features: WiFi, BLE
Crystal is 40MHz
MAC: 24:58:7c:5b:4c:18
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00310000 to 0x003effff...
Compressed 917504 bytes to 2933...
Writing at 0x00310000... (100 %)
Wrote 917504 bytes (2933 compressed) at 0x00310000 in 3.2 seconds (effective 2263.0 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
====================================================== [SUCCESS] Took 20.80 seconds ======================================================

Environment                 Status    Duration
--------------------------  --------  ------------
heltec_wifi_lora_32_v3_TNC  SUCCESS   00:00:20.804
====================================================== 1 succeeded in 00:00:20.804 ====================================================== 
 *  Terminal will be reused by tasks, press any key to close it.

richonguzman, 2024-08-22:
after each flash of fimware or filesystem (press rst) and see what data it outputs over serial. (the plug symbol of vscode)

GFaza16550, 2024-08-22:
After these two actions, i disconnected and connected again:

After pressing reset:

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x9 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x4bc
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a0c
entry 0x403c98d0

GFaza16550, 2024-08-22:
Let's try flashing again and pressing reset

GFaza16550, 2024-08-22:
After finishing successfully the firmware, it updated the screen very fast, reconnected to COM12 and showed:

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x9 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x4bc
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a0c
entry 0x403c98d0

richonguzman, 2024-08-22:
> Let's try flashing again and pressing reset

are your printed message the copy from the Serial Output of VSCODE (the one that runs when pressing the plug symbol ) ??

GFaza16550, 2024-08-22:
After trying to flash the file system, without disconnecting and connecting it again it showed:

Serial port COM12

A fatal error occurred: Could not open COM12, the port doesn't exist
*** [uploadfs] Error 2
====================================================== [FAILED] Took 13.68 seconds ======================================================

Environment                 Status    Duration
--------------------------  --------  ------------
heltec_wifi_lora_32_v3_TNC  FAILED    00:00:13.683
================================================= 1 failed, 0 succeeded in 00:00:13.683 ================================================= 

 *  The terminal process "C:\Users\033462613\.platformio\penv\Scripts\platformio.exe 'run', '--target', 'uploadfs', '--environment', 'heltec_wifi_lora_32_v3_TNC'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it

richonguzman, 2024-08-22:
> After trying to flash the file system, without disconnecting and connecting it again it showed:
>
> Serial port COM12
>
> A fatal error occurred: Could not open COM12, the port doesn't exist *** [...]

restart VSCODE and/or put esp32s3 board into boot mode (USER + RST pressed and released))

GFaza16550, 2024-08-22:
I made that and then let me flash the file system:

Serial port COM12
Connecting....
Chip is ESP32-S3 (revision v0.2)
Features: WiFi, BLE
Crystal is 40MHz
MAC: 24:58:7c:5b:4c:18
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00310000 to 0x003effff...
Compressed 917504 bytes to 2933...
Writing at 0x00310000... (100 %)
Wrote 917504 bytes (2933 compressed) at 0x00310000 in 3.3 seconds (effective 2235.2 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
====================================================== [SUCCESS] Took 20.41 seconds ======================================================

Environment                 Status    Duration
--------------------------  --------  ------------
heltec_wifi_lora_32_v3_TNC  SUCCESS   00:00:20.411
====================================================== 1 succeeded in 00:00:20.411 ====================================================== 
 *  Terminal will be reused by tasks, press any key to close it.

GFaza16550, 2024-08-22:
Then, again same behaviour:

Please build project in debug configuration to get more details about an exception.
See https://docs.platformio.org/page/projectconf/build_configurations.html

         Connected!
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x9 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x4bc
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a0c
entry 0x403c98d0

GFaza16550, 2024-08-22:
Excuse me, i'm not trying to bothering you! :) :) :)

richonguzman, 2024-08-22:
> Excuse me, i'm not trying to bothering you! :) :) :)

I know!!! we are trying to make this work! ;)

richonguzman, 2024-08-22:
porque sale a veces :

heltec_wifi_lora_32_v3_TNC

y otras 

heltec_wifi_lora_32_v3_TNC_915?

GFaza16550, 2024-08-22:
Where did you see _915? I'm not seeing it in the logs.

richonguzman, 2024-08-22:
there are two versions for TNC which you should choose according to the board.

the code you tell is not 915 although you mentioned 915 in the photos uploaded

im confused

GFaza16550, 2024-08-22:
My board is for 915 MHz, and in the platform options to upload the file system, i'm only finding HELTEC V3 TNC (without 915).

With 915 i see:

ttgo_t_lora32_v21_915_TNC

Should i use that?

richonguzman, 2024-08-22:
can you flash it with the igate firmware? to test?
                                                                                           
you where right ! I mistakenly  thought of different TNC values   board

GFaza16550, 2024-08-22:
As you requested it. :)

[image]
[image]

GFaza16550, 2024-08-22:
I think my posts should be moved to a new topic (HELTEC v3 915 MHz) :)

richonguzman, 2024-08-22:
> I think my posts should be moved to a new topic (HELTEC v3 915 MHz) :)

lets try to solve this , then we make it cleaner in github hahaha

So, it flashes right the igate firmware, now we have to get back for Tracker as TNC.

message me through "TELEGRAM" to the same user I have here @richonguzman to get it step by step until we solve it.

closed 2024-10-01 by richonguzman