richonguzman/LoRa_APRS_Tracker #136: Heltec Wireless Tracker

iw9grl, 2024-07-04:
Hello

I have an issue with the Heltec wireless tracker, if it is in power-off mode, it doesn't power-on after power-supply is connected, the orange led turn on but the board no .
Is it possible to force the power-on of the board after supply connected ? 

thank you

jclifford73, 2024-07-04:
Hello IW9GRL & Ricardo

I am using the Heltec wirless tracker and it powers on and off by removing/reconnecting USB power as long as you do NOT have a battery connected.  I use it this way in my vehicle and it works perfectly.  

I hope this help.

73 John GW4BVE

iw9grl, 2024-07-05:
> Hello IW9GRL & Ricardo
> 
> I am using the Heltec wirless tracker and it powers on and off by removing/reconnecting USB power as long as you do NOT have a battery connected. I use it this way in my vehicle and it works perfectly.
> 
> I hope this help.
> 
> 73 John GW4BVE

Hi

thank you for your answer, do you mean if I remove the 18650 battery inside it power-on when power come from usb ?
I'm using Heltec Wireless tracker in my motorcycle and it is very annoying every time remove the seat to press the reset button . 
Waiting a better solution from Ricardo ( that I thank a lot for his huge job ) I will proceed in this way, changing powering from my motorcycle , avoiding power not shutdown when I remove starter-key !

have a nice day
73 de IW9GRL

jclifford73, 2024-07-05:
Good morning IW9GRL

This is the board I use:
https://www.aliexpress.com/item/1005006966037417.html

I have it powered by USB in my vehicle with no battery connected to the board.   It powers on when I start the vehicle and off when I stop, which is as I think you want to operate. 

Have you tried removing the battery and just using USB power?

73 John GW4BVE

richonguzman, 2024-07-05:
there is no way to have a physical or code switch to turn it completely on and off with this board

so unless we have a pin checking for voltage there is no way to make it on after deep sleep

ON7MHZ, 2024-07-08:
Same Heltec Xireless tracker board here. 
After some bootloop issues ( which were solved by re-uploading file system) i have now the issue that this board does not appear on my bluetooth list on my phone.
Any suggestions ?

richonguzman, 2024-07-08:
> Same Heltec Xireless tracker board here. After some bootloop issues ( which were solved by re-uploading file system) i have now the issue that this board does not appear on my bluetooth list on my phone. Any suggestions ?

on Apple Phones you add boards in the app, not on BT over configuration of the iPhone

aprs.fi apps and others on IOS have the search for BLE devices inside it

greghall-au, 2024-07-16:
Heltec Wireless Tracker V1.1
I have loaded current version into 2 of these.
They work well (after also uploading SPIFFS;), but no Bluetooth on either.
Load same config onto T-Beam, BT is fine?
How to find the problem?
-
BTW: Thanks so much for bringing APRS into 21st Century:)

DL7AIR, 2024-07-16:
If have the same problem as greghall-au. No Bluetooth with Heltec wireless tracker V1.1
vy 73 de Klaus, DL7AIR

richonguzman, 2024-07-16:
@greghall-au and @DL7AIR , 

Heltecs have BLE (not classic BT) so (for now) APRSDROID does not have a serial protocol to work with BLE modules

it works for BLE Terminal App in Android. And also APRS.fi app on iOS APPLE phones

closed 2024-10-01 by richonguzman