richonguzman/LoRa_APRS_Tracker #138: Startup loop when using SH1106 OLED with T-Beam V1.2

MILUX-DE, 2024-07-07:
I got a hint about problems when using a T-Beam V1.2 in combination with a SH1106 OLED. So I connected both OLED types (SH1106 and SSD1306) to the I²C bus in parallel and did several compiler runs with alternating undefine of SSD1306, i.e. switching between Adafruit SSD1306 and Adafruit SH110X libraries. Each time SSD1306 was used everything went on as usual on the SSD1306-OLED, but switching to SH110X sent the T-Beam V1.2 into a loop just showing the startup screen on the SH1106 OLED.

richonguzman, 2024-07-08:
Hi,

the SH1106 cant run with the BLUETOOTH library for now, I haven't dig more for a while now to solve this. But will return to it soon

meanwhile if you can avoid BT with SH1106

radio-Joe, 2024-07-08:
Hi,

for my T-Beam V1.2 and SH1106 the boot loop disappears when correct the source code in display.cpp:
if (!display.begin(0x3c, true))  change to:  if (!display.begin(0x3c, false)) 
The OLED has no Reset, therefore false is the correct parameter.

vy73

richonguzman, 2024-07-09:
> Hi,
> 
> for my T-Beam V1.2 and SH1106 the boot loop disappears when correct the source code in display.cpp: if (!display.begin(0x3c, true)) change to: if (!display.begin(0x3c, false)) The OLED has no Reset, therefore false is the correct parameter.
> 
> vy73

Hi,

does this work even when BT is enable/active?

if YES, please share the code as where to add the change ;)

radio-Joe, 2024-07-09:
Hi Ricardo,

mni tnx for your work on iGate/Tracker fw.  :-)
My T-Beam V1.2 with SH1106 run with BT active, after modification of display.cpp. 

[INFO][AXP2101] init done!
[INFO][Main] RichonGuzman (CA2RXU) --> LoRa APRS Tracker/Station
[INFO][Main] Version: 2024.07.05
[INFO][LoRa] Initializing SX127X ...
[INFO][LoRa] LoRa init done!
[INFO][Bluetooth] Bluetooth init done!
[INFO][Main] Setup Done!
[INFO][Bluetooth] Client connected !

I am badly versed with GITHUB, so don't know how to share the corrected code! It is on line 105 (VSCODE): 
if (!display.begin(0x3c, true))  that has to be false instead of true, because there is no Reset line. 
Think you will find it. 

vy 73, dj3gi

richonguzman, 2024-07-09:
> Hi Ricardo,
>
> mni tnx for your work on iGate/Tracker fw. :-) My T-Beam V1.2 with SH1106 run with BT active, after modification of display.cpp.
>
> [INFO][AXP2101] init done! [INFO][Main] RichonGuzman [...]

all works fine if you change "true" to "false"?

the ecoMode to turn the screen off after a while and all?

radio-Joe, 2024-07-09:
also eco-mode works fine!

richonguzman, 2024-07-09:
> also eco-mode works fine!

great news.... will mod it right away

closed 2024-07-09 by richonguzman