richonguzman/LoRa_APRS_Tracker #141: Feature request: Real time in received packets logs

Tapkin12ru, 2024-07-20:
Hello!
Can you add real time settings (use network time for example) and write logs with it?

richonguzman, 2024-07-20:
Would you please explain each topic with more detail?

Tapkin12ru, 2024-07-20:
I mean this section here. Currently, the time of the received package is counted from the moment the device is loaded. It would be much more convenient if the records were marked with a real date and local network time (wifi) or GMT time.
[image]

Tapkin12ru, 2024-07-20:
As a local network time, I mean, for example, this setting in the phone
[image]

richonguzman, 2024-07-20:
This is a request for the igate firmware then , right?

Tapkin12ru, 2024-07-20:
Yes, you're right.

richonguzman, 2024-08-02:
will close it here and try to add NTP server time on igate to solve this

closed 2024-08-02 by richonguzman