richonguzman/LoRa_APRS_Tracker #143: BT Not working on Supreme V3 - Lilygo

snag1es, 2024-08-02:
I think the BT is not working. 
When upload firmware with value persistent active the Phone does not recognize the device when searching for new BT connection. When enter config menu the BT is in state OFF. When enable it via the menu the BT did not start again. Trying to debug using serial terminal and serial log. but only appeared messages received directly from the device.

richonguzman, 2024-08-02:
Hi,

ESP32S3 (in Tbeam Supreme) only does BLE not BT

how are you looking for it? on which phone?

snag1es, 2024-08-05:
Hi,
I'm using CAT S 75 S and trying the APRS Droid app to connect.
I have one more same hardware with another firmware using 868 Meshtastic. The device communicates with the Meshtastic app without any problems using BLE.
The problem is that I cannot establish any connection even pairing with the device. It was not recognizable in the air.

richonguzman, 2024-08-05:
hi,

the problem is on APRSDROID side as they have not BLE support yet

snag1es, 2024-08-06:
I see, OK thanks for the explanation.
We can close this case then.

closed 2024-08-06 by richonguzman