richonguzman/LoRa_APRS_Tracker #146: Button on custom build

RobertMavric, 2024-08-21:
I made my tracker with TTGO_T_LORA32_V2 and I want to add button (GPIO4)+ GPS with 115200 without success.
In "boards_pinout.h" under "TTGO_T_LORA32_V2_1_GPS) I added:
```
 #define BUTTON_PIN 4 //Button to GND
 #define HIGH_GPS_BAUDRATE //GPS with 115200
```

In the main program I can see it working with the One button library with the pullup resistor on and connected to GND. I also tried with an additional resistor, but it doesn't work.
```
#ifdef BUTTON_PIN
 OneButton userButton = OneButton(BUTTON_PIN, true, true);
```

Any idea what I'm doing wrong?

73 S56RT

richonguzman, 2024-08-21:
#define BUTTON_PIN 4 

was this added on boards_pinout.h on line 61 to 77 or in between?

this should validate the button to work

please confirm fist the button part

RobertMavric, 2024-08-29:
Apologies for the late reply.
The button works but is sometimes unresponsive. Probably bad quality of my keys ;)