richonguzman/LoRa_APRS_Tracker #147: Shutdown through wifi or bluetooth

iw9grl, 2024-08-21:
Hi

I don't know if it is possible to do, I installed a tbeam in my motorcycle in a very hidden place :)  and I need to shutdown the tbeam board when it 's parked in my own garage ( no GPS available, but I want to save battery charge ) .
Is it possible through bluetooth or wifi, run a shutdown ? I think, could be usefull the possibility to run a local webserver like the iGate ones , to configure all parameter of the tbeam and, why not, have the shutdown button.

thank you
best 73
IW9GRL

richonguzman, 2024-08-21:
I think on a few solutions:

- web server (more consumption)
- external powerOff button somewhere on the motorbike
- special code to be sended over BT to the tracker?

Mane76, 2024-08-21:
Maybe oldschool, 

but is there a possibility to get voltage on a motorcycle only when ignition is on (like in cars)?
If yes and when you remove the battery from the tracker it would automatically be turned on and off.

iw9grl, 2024-08-22:
@Mane76 the idea is to leave battery on the tracker to have it active when motorbike is off (outside home) actually the usb power is under key management, when motorbike is off no power on usb. The problem is, if I leave motorbike in garage (no need to have tracker), the battery discharge and with a new power I don't have enough time to charge it, for example in an half hour (when I move to the office in the morning).
@richonguzman if no other solution is possible I will put an external button to disconnect power, but the idea to have code over BT could be great!
Tnx
73

iw9grl, 2024-08-22:
In the meantime I think another solution could be really interesting, I don't know if you already put in place the shutdown after usb power is off ( I see some other requests from other guy), could be really interesting to wake up the board with a shake sensor connected in some GPIO active only during the deep sleep (that connect gnd or vdd to a GPIO pin), enable or not through the configuration file, in this way we don't need any other external switch :) is it a crazy request?? 😂😂

richonguzman, 2024-08-22:
> is it a crazy request?? 😂😂

yes it is! hahahaha

but will see what I can do as this is a very specific situation to solve

iw9grl, 2024-08-22:
The only things I need just for summary, is to poweroff board remotely if tracker is not accessible where installed

richonguzman, 2024-08-22:
> The only things I need just for summary, is to poweroff board remotely if tracker is not accessible where installed

like receiving a special message over aprs? 

because BT does not have much range for remote turning it of

iw9grl, 2024-08-22:
I need to turn off very close to the tracker

richonguzman, 2024-10-01:
not a clean solution for now, but I dream on building an app that controls Tracker from phone

closed 2024-10-01 by richonguzman