richonguzman/LoRa_APRS_Tracker #148: Unable to load to T-Beam AXP2101-V1.2

craigsned, 2024-08-22:
I'm unable to load the config file to T-Beam APX2101-V1.2.   The firmware loads OK.

I've been able to load alternative APRS Tracker software using the same method - VSCode - Upload firmware - upload config and works fine.   :-(  The alternative does list the T-Beam APX2101-V1.2. as a specific environment.  I would have thought that the T-Beam V1.2 would have worked for this application, but doesn't seem to.

Also of note, I have been able to upload your APRS Gateway using the web installer to the same T-Beam APX2101-V1.2 without any issues.

richonguzman, 2024-08-22:
how does it fail to upload the filesystem?

error on vscode output? or problem in editing the tracker_conf.json file?

craigsned, 2024-08-22:
No change in tracker display and not beaconing (or doesn't seem to be).  I had expected the display to show my callsign/ssid.

richonguzman, 2024-08-22:
but you see LORA APRS welcome message in display and nothing else?

craigsned, 2024-08-22:
Yes, nothing on the display changes - stays the same as the photo (not my photo, not my screen, but this is what is displayed).  

[image]

richonguzman, 2024-08-22:
ok, does you /data/Tracker_conf.json file has any small red waves in it ? (anywhere?)

craigsned, 2024-08-22:
Not that I can see..

{
	"beacons": [
		{
			"callsign": "MM0NBW-15",
			"gpsEcoMode": false,
			"symbol": "[",
			"overlay": "/",
			"micE": "",
			"comment": "LoRa Tracker",
			"smartBeacon": {
				"active": true,
				"slowRate": 120,
				"slowSpeed": 3,
				"fastRate": 60,
				"fastSpeed": 15,
				"minTxDist": 50,
				"minDeltaBeacon": 20,
				"turnMinDeg": 12,
				"turnSlope": 60
			}			
		},
		{
			"callsign": "MM0NBW-15",
			"gpsEcoMode": false,
			"symbol": ">",
			"overlay": "/",
			"micE": "",
			"comment": "LoRa Tracker",
			"smartBeacon": {
				"active": true,
				"slowRate": 120,
				"slowSpeed": 10,
				"fastRate": 60,
				"fastSpeed": 70,
				"minTxDist": 100,
				"minDeltaBeacon": 12,
				"turnMinDeg": 10,
				"turnSlope": 80
			}
		},
		{
			"callsign": "MM0NBW-15",
			"gpsEcoMode": false,
			"symbol": "b",
			"overlay": "/",
			"micE": "",
			"comment": "LoRa Tracker",
			"smartBeacon": {
				"active": true,
				"slowRate": 120,
				"slowSpeed": 5,
				"fastRate": 60,
				"fastSpeed": 40,
				"minTxDist": 100,
				"minDeltaBeacon": 12,
				"turnMinDeg": 12,
				"turnSlope": 60
			}
		}
	],
	"display": {
		"showSymbol": true,
		"ecoMode": false,
		"timeout": 4,
		"turn180" : false
    },
	"other": {
		"simplifiedTrackerMode": false,
		"sendCommentAfterXBeacons": 10,
		"path": "WIDE1-1",
		"nonSmartBeaconRate": 15,
		"rememberStationTime": 30,
		"maxDistanceToTracker": 30, 
		"standingUpdateTime": 15,
		"sendAltitude": true,
		"sendBatteryInfo": false,
		"bluetoothType": 1,
		"bluetoothActive": true,
		"disableGPS": false
	},
	"winlink": {
		"password": "ABCDEF"
	},
	"bme": {
		"active": false,
		"temperatureCorrection": 0.0,
		"sendTelemetry": false
	},
	"notification": {
		"ledTx": false,
		"ledTxPin": 13,
		"ledMessage": false,
		"ledMessagePin": 2,
		"ledFlashlight": false,
		"ledFlashlightPin": 14,
		"buzzerActive": false,
		"buzzerPinTone": 33,
		"buzzerPinVcc": 25,
		"bootUpBeep": false,
		"txBeep": false,
		"messageRxBeep": false,
		"stationBeep": false,
		"lowBatteryBeep": false,
		"shutDownBeep": false
	},
	"lora": [
		{
			"frequency": 43912500,
			"spreadingFactor": 12,
			"signalBandwidth": 125000,
			"codingRate4": 5,
			"power": 20
		},
		{
			"frequency": 43912500,
			"spreadingFactor": 9,
			"signalBandwidth": 125000,
			"codingRate4": 7,
			"power": 20
		},
		{
			"frequency": 439912500,
			"spreadingFactor": 12,
			"signalBandwidth": 125000,
			"codingRate4": 5,
			"power": 20
		}
	],
	"pttTrigger": {
		"active": false,
		"io_pin": 4,
		"preDelay": 0,
		"postDelay": 0,
		"reverse": false
	}
}

richonguzman, 2024-08-22:
I pasted the code here to check and all looks good:

now into the (once again) upload filesystem part

so :
1) turn off VSCODE
2) disconnect Tracker from usb from PC

turn on VSCODE again

connect Tbeam with usb cable to PC

and just then , try to "UPLOAD FILESYSTEM IMAGE" and return with comments ;)

craigsned, 2024-08-22:
Thanks - uploaded filesystem image only and no change.  Here is a paste of the output from VSCODE:

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/ttgo-t-beam.html
PLATFORM: Espressif 32 (6.7.0) > TTGO T-Beam
HARDWARE: ESP32 240MHz, 1.25MB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20016.0 (2.0.16)
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - tool-mkfatfs @ 2.0.1
 - tool-mklittlefs @ 1.203.210628 (2.3)
 - tool-mkspiffs @ 2.230.0 (2.30)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ deep+, Compatibility ~ soft
Found 53 compatible libraries
Scanning dependencies...
Dependency Graph
|-- RadioLib @ 6.6.0
|-- TinyGPSPlus @ 1.0.3
|-- Time @ 1.6.1
|-- OneButton @ 1.5.0
|-- ArduinoJson @ 6.21.3
|-- NimBLE-Arduino @ 1.4.1
|-- esp-logger @ 1.0.0
|-- Adafruit BusIO @ 1.16.1
|-- Adafruit Unified Sensor @ 1.1.14
|-- Adafruit BME280 Library @ 2.2.4
|-- Adafruit BMP280 Library @ 2.6.8
|-- Adafruit BME680 Library @ 2.0.4
|-- XPowersLib @ 0.2.4
|-- Adafruit GFX Library @ 1.11.9
|-- Adafruit SSD1306 @ 2.5.10
|-- Adafruit SH110X @ 2.1.10
|-- BluetoothSerial @ 2.0.0
|-- WiFi @ 2.0.0
|-- APRSPacketLib
|-- FS @ 2.0.0
|-- Wire @ 2.0.0
|-- SPI @ 2.0.0
|-- KISS_TO_TNC2
|-- SPIFFS @ 2.0.0
Building in release mode
Building FS image from 'data' directory to .pio\build\ttgo-t-beam-v1_2\spiffs.bin
/tracker_config.json
Looking for upload port...
Auto-detected: COM7
Uploading .pio\build\ttgo-t-beam-v1_2\spiffs.bin
esptool.py v4.5.1
Serial port COM7
Connecting.........
Chip is ESP32-D0WDQ6-V3 (revision v3.1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 08:f9:e0:f7:50:6c
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00310000 to 0x003effff...
Compressed 917504 bytes to 2925...
Writing at 0x00310000... (100 %)
Wrote 917504 bytes (2925 compressed) at 0x00310000 in 4.9 seconds (effective 1512.8 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
========================================================= [SUCCESS] Took 23.02 seconds =========================================================

Environment       Status    Duration
----------------  --------  ------------
ttgo-t-beam-v1_2  SUCCESS   00:00:23.020
========================================================== 1 succeeded in 00:00:23.020 ==========================================================
 *  Terminal will be reused by tasks, press any key to close it.

richonguzman, 2024-08-22:
now,  with the board connected (and VSCODE open) press on the bottom the "plug" symbol 

the energy plug connector (this will open the serial port info)

wait till this new window is ready in viscose and press RST button on the tbeam
this should give us info on the booting process

craigsned, 2024-08-22:
OK, not sure what exactly fixed it, but here is what I did;
Downloaded the source files again.
Edited the config json file.
Uploaded firmware
Uploaded config

**At this stage nothing happened.

Then I saw your message about pressing the "plug" button on VSCode - which I did and now the tracker seems to be working ok...

Strange behaviour, but we got there!  Thank you for your help with this!  It's very much appreciated!

Best,
Craig

richonguzman, 2024-08-22:
> OK, not sure what exactly fixed it, but here is what I did; Downloaded the source files again. Edited the config json file. Uploaded firmware Uploaded config
>
> **At this stage nothing happened.
>
> Then [...]

FINALLY WE SUCCEED!!!!!   great!!

now enjoy! :)

closed 2024-08-22 by richonguzman