richonguzman/LoRa_APRS_Tracker #149: missing SConscript file ‘C:\Users\.platformio\\\packages\framework-arduinoespressif32\\tools\platformio-build.py

IU2IGX, 2024-08-27:
Hi,

I have just come into possession of a ‘Heltec Wireless Tracker’ and wanted to programme it. I followed your instructions but unfortunately I always get the error when compiling:

*** missing SConscript file ‘C:\Users\.platformio\\\packages\framework-arduinoespressif32\\tools\platformio-build.py’
File ‘C:\Users\Utente\.platformio\platforms\espressif32\builder\frameworks\arduino.py’, line 36, in <module>.

Unfortunately, google is not much help, so I wanted to ask if anyone else had experienced the same problem, as it is totally blocking.

Thanks & Best Regards,

Daniel/IU2IGX
[image]

richonguzman, 2024-08-27:
Hi, which Operating System are you running?
is VSCODE up to date?
did you import the project(unzipped folder) or did you just open it?

IU2IGX, 2024-08-28:
Hi Richonguzman,

thank you for your reply! 
OS: Win10 Pro
VSCode: 1.92.2 (2024-08-14)
PlatformIO IDE: 3.3.3
The project was imported from the zip archive extracted into the folder as indicated in the guide.

Thanks for the help!

richonguzman, 2024-08-28:
change the expressif version from 6.7.0 to 6.3.1 in platformio.ini , save it and test the full upload of data

IU2IGX, 2024-08-29:
Hi,

the correction solved the problem concerning the SConscript file.
Compilation now takes place correctly and so does the upload to the Tracker (HTIT-Tracker V1.1), but I ran into the error that the Tracker restarted in a loop.

IU2IGX, 2024-08-29:
OK, after a few attempts to load it now works!
Thanks Ricardo! Your projects look great!

closed 2024-08-29 by IU2IGX