richonguzman/LoRa_APRS_Tracker #152: Heltec Wireless Tracker module boot cycle problem

g3cwi, 2024-09-04:
A number of peope are encoutering problems with the HWT. The problem manifests itself with the tracker continually trying to boot. Having set up and used many of these devices for SOTA people, I can offer some insight into the problem - but not the cure. Various people have suggested that they have "fixed it" by altering settings but I don't think they have - it just hasn't happened again yet. The other day I did a clean flash of the firmware on a new device to try some things out. I have observed that the problem :

- only occurs after the device has been switched on and off numerous times (this is why people think they have fixed it)
- that the problem starts with a single boot cycle crash
- that the problem gets progressively worse until it can take an hour or more of boot cycling until the unit boots properly
- that the unit will always boot properly (eventually)
- the unit sometimes seems to get futher through the boot cycle and the second screen (with callsign etc) appears momentarily. Mostly this is not seem though and the crash occurs after the first screen

I am using the latest firmware but the older pulls had the same problem. The trackers are set to BLE (either IOS or Android). In my case, the display is set to power down after 90 seconds.

I have seen this problem on 4 different HWT modules and I expect it happens to all of them eventually.

I did experiment with adding some delays in the boot up sequence but these made no difference. I also switched Bluetooth off altogether but the problem still occured. It can take 50 switch or more switch on sequences over a few days for the problem to start.

Mane76, 2024-09-04:
Hello Richard,
this is interesting to hear.
We also use some HWTs without any issue, I also use one personally in the car with ON-OFF cycles incl, the usage of BLE connected with android.

Just some question: 

- Which OS are you using to flash the firmware ? 
- Which Version of VSCode ? 
- Do you have chosen the right board in Firmware AND Filesystem ?
- Did you consider to press the USER Button on the HWT prior connecting to USB before flashing ?
- When the reboot happens, which screen is shown ?
- When the HWT makes it to the 2nd screen with callsign, is the everything fine or do also reboots happen ?

BR - Mane

g3cwi, 2024-09-04:
Hi Mane

Which OS are you using to flash the firmware ?
**_Windows 10_**
Which Version of VSCode ?
**_1.92.2_**
Do you have chosen the right board in Firmware AND Filesystem ?
**_Yes_**
Did you consider to press the USER Button on the HWT prior connecting to USB before flashing ?
**_Yes this is how I do it_**
When the reboot happens, which screen is shown ?
**_The standard startup screen_**
When the HWT makes it to the 2nd screen with callsign, is the everything fine or do also reboots happen ?
**_No, the reboot can still happen. The 2nd screen is only seen for a fraction of a second before the reboot._**

Mane76, 2024-09-04:
I sent you a mail on the address shown at qrz.com.

richonguzman, 2024-09-04:
hi  @Mane76 @g3cwi 

can you please share the tracker_conf.json for further analysis ??  this is so strange as it never happened to me once

g3cwi, 2024-09-04:
It is very odd, but I others have seen this in the UK too. It takes a long while to show itself.

[tracker_config.json](https://github.com/user-attachments/files/16870324/tracker_config.json)

g3cwi, 2024-09-04:
Okay, thanks. The fork from Mane is as standard with all the frequencies. I have set the UK frequency by the main menu. Let's see if that's the problem.

richonguzman, 2024-09-08:
hi @g3cwi , how are the test going?

g3cwi, 2024-09-08:
> hi @g3cwi , how are the test going?

No problems so far. The unit has been switched on and off several times each day and used extensively. If it is still okay by Wednesday I will reflash it with the BLE enabled. Someone on the SOTA reflector said that in his case, the reboot problem was cured when the BLE/Bluetooth was disabled. Thanks for your interest!

g3cwi, 2024-09-12:
Update. I ran the tracker without BLE enabled without any problems. Today I will reflash with Mane's fork but this time I will enable BLE and display timeout as per my usual useage. I will report back in a week or so.

g3cwi, 2024-09-21:
Yesterday the boot cycle problem reoccurred on switch on. I connected the tracker to USB power and eventually it booted correctly. Ideas on further testing welcome.

Jumpy07, 2024-09-24:
Having the same issue.. tried disabling BT but no luck.. keeps rebooting at first screen.. anyone figure this out ?

richonguzman, 2024-09-24:
Which Firmware are you using ? @Jumpy07 and @g3cwi now?

what BluetoothType value are you saving on tracker_conf?

Jumpy07, 2024-09-24:
Tried 1 and O for Bluetooth and  have also tried disabling it. setting false.

I get the Lora APRS Tracker splash screen.. then it reboots..

I have also tried different versions of Espressif, currently on 3.7.0

Also tried different power supplies..

richonguzman, 2024-09-24:
> Tried 1 and O for Bluetooth and have also tried disabling it. setting false.
> 
> I get the Lora APRS Tracker splash screen.. then it reboots..
> 
> I have also tried different versions of Espressif, currently on 3.7.0
> 
> Also tried different power supplies..

You have Heltec Wireless Tracker, right?
this has ESP32S3 and this board has ONLY BLE, so define it as "bluetoothType": 0 (or 2 but this is only for developers)
the bluetoothActive will be fixed to ON for now as it needs a few more tests.

the expressif version should not affect at all with this reboots problem

did you upload both new firmware and new filesystem? (sometimes updates change filesystem also and when tracker does not find an old variable it could crash)

Jumpy07, 2024-09-24:
Yes, Heltec Wireless Tracker V1.1

Have tried bluetoothtype 0

Flashed it with meshtastic and was working fine.. also flashed igate and was fine..   .. only when flashed with tracker via VSC does it go into a reset loop after splash screen..

Will try flashing new firmware and see what happens..

richonguzman, 2024-09-24:
> Yes, Heltec Wireless Tracker V1.1
> 
> Have tried bluetoothtype 0
> 
> Flashed it with meshtastic and was working fine.. also flashed igate and was fine.. .. only when flashed with tracker via VSC does it go into a reset loop after splash screen..
> 
> Will try flashing new firmware and see what happens..

remember to flash BOTH PARTS: firmware and filesystem

g3cwi, 2024-09-25:
> Which Firmware are you using ? @Jumpy07 and @g3cwi now?
> 
> what BluetoothType value are you saving on tracker_conf?

Hi Ricardo. I am currently using @Mane76 's Pull. Bluetooth set for iPhone BLE (type 0). Happy to try other configurations.

richonguzman, 2024-09-25:
> > Which Firmware are you using ? @Jumpy07 and @g3cwi now?
> > what BluetoothType value are you saving on tracker_conf?
> 
> Hi Ricardo. I am currently using @Mane76 's Pull. Bluetooth set for iPhone BLE (type 0). Happy to try other configurations.

how is It working for you?

Jumpy07, 2024-09-30:
Finally got mine sorted.. was issues with board filesystem.. worked third time lucky..

richonguzman, 2024-09-30:
> Finally got mine sorted.. was issues with board filesystem.. worked third time lucky..

great news!

richonguzman, 2024-10-01:
asuming all works now this closed for soon to be released WebFlasher ;)

closed 2024-10-01 by richonguzman