richonguzman/LoRa_APRS_Tracker #153: APRS.fi on iPhone 11 does not Establish BLE Connection with Tracker on T-Beam v1.1

khkremer, 2024-09-05:
I have the latest tracker software installed. Here is the version information from the bottom of the configuration page:

ttgo-lora32-v21
-DTTGO_T_LORA32_V2_1,-DHAS_SX1278,-DHAS_ADC_CALIBRATION
Version date: 2024.08.14
Build date: 2024-08-14 18:25:40 UTC

I have the current version of APRS.fi (2.5.3.4) on my iPhone 11. When I go to the TNC setup, it just spins while looking for a BLE TNC connection. 

I've tried this with both BT Android set to on and off (just in case "normal" BT would interfere with BLE), without a change in behavior. 

Is there anything else I need to configure to get this working?

richonguzman, 2024-09-05:
Are you using bluetoothType: 0 ?

This is the one for apple ble and aprs.fi app

khkremer, 2024-09-05:
Good question :) Thanks for reminding me. Now that you mentioned, it, I remember reading this, but that was before I decided to give BLE a try, so I just followed the Wiki about BT/BLE connectivity, and that did not mention the bluetoothType, so I completely forgot about it. 

Sorry for not RTFM. It is all working now.

Thank you!
de Karl Heinz - K5KHK

closed 2024-10-01 by richonguzman