richonguzman/LoRa_APRS_Tracker #155: “bluetoothType”: 1 not working on heltec wireless tracker

9M2PJU, 2024-09-12:
rebuilding and reflashing few times and it did not showing up. “bluetoothType”: 0 is up but I dont have iPhone to test.

richonguzman, 2024-09-12:
> rebuilding and reflashing few times and it did not showing up. “bluetoothType”: 0 is up but I dont have iPhone to test.

as per WIKI info:
"bluetoothType":1 is for Android and BT classic

all ESP32S3 based boards have BLE (so "bluetoothType":0 (for iPhone + aprs.fi) or "bluetoothType":2 for Android and BLE Terminal for testing right now))

9M2PJU, 2024-09-12:
> as per WIKI info: "bluetoothType":1 is for Android and BT classic
>
> all ESP32S3 based boards have BLE (so "bluetoothType":0 (for iPhone + aprs.fi) or "bluetoothType":2 for Android and BLE Terminal [...]

Thanks, I will try type 2 and wiki should include type 2 too.

richonguzman, 2024-09-12:
> Thanks, I will try type 2 and wiki should include type 2 too.

it will once we know of an app to make this work as it should

9M2PJU, 2024-09-12:
[image]

Yes, maybe APRSDroid developer should notice.

Thanks again.

richonguzman, 2024-09-12:
> ![image](https://private-user-images.githubusercontent.com/991353/366953966-09cd9243-1dba-4c10-a924-70c3e8c28438.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3Lm [...]

they already know:

there is no BLE serial packet format defined, this is why they dont have a BLE available version (for now)