richonguzman/LoRa_APRS_Tracker #156: DIY Tracker on LLCC68 doesn't start

Majakkf, 2024-09-24:
Hi,
I recently built a diy tracker on esp32 and Ebyte E220 400M30S (LLCC68)
The problem is that it doesn't start, I keep getting this error:
Starting LoRa failed! State: -2
Everything is wired right
What is the possible solution?

richonguzman, 2024-09-24:
What are your Lora settings? Sf12?

Majakkf, 2024-09-25:
> What are your Lora settings? Sf12?

[image]
I am using SF9

richonguzman, 2024-09-25:
> > What are your Lora settings? Sf12?
> 
>  SF9

power is on 18 (valid power units are 17, 20, 22... for your board max is 20). change it!

Majakkf, 2024-09-25:
I changed it to 17 and it still doesn't start

richonguzman, 2024-09-25:
the [RADIOLIB_ERR_CHIP_NOT_FOUND]   (-2) you experience is because it can recognize the Lora Module

please check wiring / pins

Majakkf, 2024-09-26:
Wiring is good and it still doesn't start

richonguzman, 2024-10-01:
did you solve this?

Majakkf, 2024-10-02:
No, maybe Lora module is dead?

richonguzman, 2024-10-02:
could you test it with other firmwares?

richonguzman, 2024-10-02:
> What other firmwares are there?

is there a meshtastic one for this ESP32 + LLCC68 board?

Majakkf, 2024-10-04:
> > What other firmwares are there?
> 
> is there a meshtastic one for this ESP32 + LLCC68 board?

Yes, meshtastic starts on LLCC68 (RA-02SC) but not on E220 400M30S which is also on LLCC68 chip 
[image]

richonguzman, 2024-10-04:
I received confirmation of another Ham OP which used ESP32 DIY with LLCC68 board with success

So I would suggest to double check the connections, the esp32 boards and finally see if E220 boards is working

richonguzman, 2024-10-29:
did you try latest update of tracker?

harifhaqim, 2025-02-16:
tried with RA-01SC(LLCC68), removed the 1W definition but still...same issue with OP

richonguzman, 2025-02-16:
> tried with RA-01SC(LLCC68), removed the 1W definition but still...same issue with OP

1W definition is needed!

I heard lots of successful stories about using this chip, but! is your board the 1W version or not? they have different pins!

harifhaqim, 2025-02-16:
not a 1W chip, it's RA-01SC with LLCC68...if it does need 1W version, the original OP would have been succeeded..and both 1W and 0.1W version have the same library..

richonguzman, 2025-02-16:
so not found...  maybe have to check on the pins definition

harifhaqim, 2025-02-16:
i have also made a remap

DIO1 -> 4
BUSY -> 2
RXEN -> 1(TX0)
TXEN -> 3(RX0)

already appended to the code, check continuity and it's fine, the RadioLib said it might be NSS(CS) wrong pin but I've tested it and using the default GPIO

richonguzman, 2025-02-16:
first, what are the pins used for this lora unit?

harifhaqim, 2025-02-16:
#define RADIO_SCLK_PIN      18
    #define RADIO_MISO_PIN      19
    #define RADIO_MOSI_PIN      23
    #define RADIO_CS_PIN        5
    #define RADIO_RST_PIN       27
    #define RADIO_DIO1_PIN      12
    #define RADIO_BUSY_PIN      14
    #define RADIO_RXEN          32
    #define RADIO_TXEN          25

this is the stock pinout not including the remap.

richonguzman, 2025-02-16:
I know but can you link the product data sheet to see if we really need that much pins defined or maybe check if there is a new different needed

harifhaqim, 2025-02-16:
this is the module for Ra-01SC I'm using 

[image]

harifhaqim, 2025-02-16:
> I know but can you link the product data sheet to see if we really need that much pins defined or maybe check if there is a new different needed

[ra-01sc_specification.pdf](https://github.com/user-attachments/files/18814931/ra-01sc_specification.pdf)

richonguzman, 2025-02-16:
please the data sheet link

I already thing there should be no TxEN and RxEN pins for your board

richonguzman, 2025-02-16:
now I've seen the data sheet

you need to remap all of them:

#define RADIO_SCLK_PIN      ?
#define RADIO_MISO_PIN      ?
#define RADIO_MOSI_PIN      ?
#define RADIO_CS_PIN        ?
#define RADIO_RST_PIN       ?
#define RADIO_DIO1_PIN     ?
#define RADIO_BUSY_PIN      ?
#define RADIO_RXEN          ?
#define RADIO_TXEN          ?

you had them defined before, but is this wired as you wrote?

harifhaqim, 2025-02-16:
i used the stock pinout with some of them remapped, so this is my current pinout

#define RADIO_SCLK_PIN      18
#define RADIO_MISO_PIN      19
#define RADIO_MOSI_PIN      23
#define RADIO_CS_PIN        5
#define RADIO_RST_PIN       27

The Remapped Pins
#define RADIO_DIO1_PIN      4
#define RADIO_BUSY_PIN      2
#define RADIO_RXEN          1
#define RADIO_TXEN          3

richonguzman, 2025-02-16:
are this pins really available on your esp32 ???

> i used the stock pinout with some of them remapped, so this is my current pinout
>
> #define RADIO_SCLK_PIN 18 #define RADIO_MISO_PIN 19 #define RADIO_MOSI_PIN 23 #define RADIO_CS_PIN 5 #define [...]

harifhaqim, 2025-02-16:
I'm using this ESP32 module 

[image]

harifhaqim, 2025-02-16:
so the TXEN and RXEN are the wrong pin for me?

richonguzman, 2025-02-16:
> so the TXEN and RXEN are the wrong pin for me?

I would test 12,13,14 for them and test

harifhaqim, 2025-02-17:
> > so the TXEN and RXEN are the wrong pin for me?
> 
> I would test 12,13,14 for them and test

I have remapped the pin to GPIO 26 for RXEN and GPIO 14 for TXEN....still -2 error

richonguzman, 2025-02-17:
I don't know what else to do with this board. It seems you've tested all the needed things to make it work

closed 2025-03-18 by richonguzman