richonguzman/LoRa_APRS_Tracker #157: Voltage as telemetry doesn't work when you change symbol

IU5HKU, 2024-09-25:
Don't know if it's my fault, but when i change symbols, as example from -9 (car) to -5(runner) the telemetry works partially, voltage is sent, but the first packets needed for the proper "formatting" of the telemetry doesn't, so you haven't a proper plot. Seems that the parms, unit, eqns and bit packets are sent only one time, when the tracker is powered on, and not when you change symbols. Maybe i'm doing something wrong?

richonguzman, 2024-09-25:
You are right!

Will reset this inicial 3 packets command for when you change between callsings

IU5HKU, 2024-09-26:
I have a proposal, i don't wanna open another issue for this, it' possible to put a line into the configuration.json for a deepsleep of the tracker when battery goes below a certain voltage? Most trackers are used with lipo batteries, and when you are around you forgot to check the tracker to power off it manually in case of low voltage.
In this way you can save the life of the battery a lot. Thank you

richonguzman, 2024-10-01:
sendTelemetry is reseted when callsign is changed now with next firmware release.

also lowBatteryVoltage watcher is added to avoid damaging batteries

closed 2024-10-01 by richonguzman