richonguzman/LoRa_APRS_Tracker #159: WifiAP - Web Configuration

iw9grl, 2024-10-02:
Hi Ricardo

I tried the Web configuration of the board ( Heltec WT )  and it works , but I'm not able to exit from WifiAP, tracker usage doesn't start

there are some particular sequences of buttons to exit from WifiAP ?

thank you
73

richonguzman, 2024-10-02:
when you save it should reboot and go right into Tracker mode, unless some data saved is not compatible with this.

any special icon symbol character used?

iw9grl, 2024-10-02:
Hi Ricardo

the icons used are man, car and motorbike \> not any other special symbols .

Another strange behaviour I noticed in Heltec WT, many times the board goes in boot loop, this not happens with previous firmware I used, if I remember well the ones at the end of August

73

richonguzman, 2024-10-02:
i will close this as new webflasher has many corrections

if you find anything let me know in detail in another"issue" please

closed 2024-10-02 by richonguzman

iw9grl, 2024-10-02:
Sure, thank you