richonguzman/LoRa_APRS_Tracker #160: Feature proposal: gps shutdown - wakeup

IU5HKU, 2024-10-02:
Hi Ricardo, first of all thank you for the lowBatteryVoltage settings, very useful indeed. This isn't an issue, i have another proposal, why not send the shutdown command to the gps module to improve the sleepmode? Ublox gps chips can be put in a "backup mode" where it consume very little current, stated below 200uA, not bad at all. Can be also waked up from this state simply writing some chars over the serial, i need only some form of "activity" to wake up and do an hot start, so it could retrieve and track satellites from the "backup" mode very quickly...i0ve see you are using TinyGPS++ as gps library, that is a generic gps library, but you can use simple serial communication to send the "sleep" command to the gps chip, i've read this many years ago...i'll try to retrieve where....here it is:

[https://forum.arduino.cc/t/gps-module-power-management/477859/15](url)

as you can read is simply matter to send an array of byte, and all the uBlox chips are compatible with this command.
Hope this can help, thank you for your time, have a nice day!

richonguzman, 2024-10-02:
this is ment to replace the current "gpsEcoMode" ? or to add this to boards without AXP2101/AXP192???

IU5HKU, 2024-10-03:
I've missed the reading of the code who uses the PMU to cut the power to the gps chip, ok, just read it, and yes, well done. Adding this feature in boards where PMU isn't present could be a great improvement, really, i think that the wast majority of us out there are using a cheapo uBlox chip (mostly in the neo6 version, is the cheapest..). 
Don't know how much time it takes to regain the proper sat communication after a cold reboot, but i think that an hot one is faster for sure, and with this command the gps chip keep in a non volatile memory the last "seen" sats, i think that the tracking in this case should be quite immediate. For my toughts, i would keep your "ecomode" as an extreme measure in case of lowBatteryVoltage trigger, when you need to shutdown everything possible to preserve battery.
So i intend it not a replacement, but an improvement when boards have a PMU, and a similar way of use in handcrafted trackers, so you don't have to do additional soldering for transistors or some sort of power cutting deviced for the gps power. If this command works as expected (i hope) there is also the pleasant side effect o improve the battery duration :-)

richonguzman, 2024-10-03:
it would be awesome to get the exacto lines/commands to make it sleep and wake as needed without turning it of

could not find both, just one, but could not test it now

IU5HKU, 2024-10-03:
This is the part of code from the link in the previous post:

void setup() {

  Serial.begin(9600);
  Serial.println( F("NeoGPS library & current reduction test!") );

  // gpsPort.begin(9600);
  GPS.send_P( &gpsPort, F("PMTK314,0,1,0,1,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0") ); // RMC+GGA
  delay(50);
  GPS.send_P( &gpsPort, F("PMTK220,1000") ); // 1Hz
  delay(50);
  GPS.send_P( &gpsPort, F("PGCMD,33,0") );   // No antenna status
  delay(5000);                               // wait 5 seconds before standby mode

  Serial.println( F("") );
  Serial.println( F("In GPS standby mode for 10 seconds...") );
  GPS.send_P( &gpsPort, F("PMTK161,0") );    // send standby mode command to GPS (current = 1.02MA)

  delay(10000);
  Serial.println( F("") );

  //GPS.send_P( &gpsPort, F("P") );          // wake module send a byte (current = 21.9MA)
  Serial.println( F("GPS awake now...") );   // or will wake on Serial.print
  Serial.println( F("") );
}

it uses the neogps library, as this commands work only for the uBlox chips, but the commands can be sent over the gps erial as well. Don't worry,  i know , the most precious resource is the time, you will add this feature if you want and if you have enough time, there's no problem at all, it's only ideas to improve this already great piece of code. I've not enough time to help you in writing code, otherwise I would have helped you for sure...so sorry about that :-(

richonguzman, 2024-10-03:
I really want to test this, do you know which library was used for this?

IU5HKU, 2024-10-03:
NeoGPS library for what i remember...

IU5HKU, 2024-10-03:
For handcrafted trackers that haven't APX chips, another solution could be to connect gps module Vcc to a free GPIO of choice on the board, usually gps draw around 20-30mA and you don't risk to burn the digital input. If you need to power off completely the module, in this way you can do it with a simple digitalWrite(GPIO,0) and with "1" you can poweron the module.

richonguzman, 2024-10-07:
well I manage to test all modes:

- Continuous Mode
- Power Save Mode
- Eco Mode
- Trickle Mode
- BackUp Mode
- StandBy Mode

some were the same as Continuous (which is the default) and the PowerSaveMode was not that good as it sleep but uses then again much current to get again gps fix. So it seems I can get a good results out of this test

IU5HKU, 2024-10-09:
Backup Mode is the best available for this purpose for what i've read around, the power consumption in mA is in the 1 to 20 order and you have the list of the seens sats available at wakeup for a fast tracking, i think you should go for this mode