richonguzman/LoRa_APRS_Tracker #161: support for ttgo-t-beam-AXP2101-v1_2

ch0mik, 2024-10-02:
plase add support for board ttgo-t-beam-AXP2101-v1_2

richonguzman, 2024-10-02:
for more than a year the boards is been supported as: ttgo-t-beam-v1_2

closed 2024-10-02 by richonguzman

ch0mik, 2024-10-03:
@richonguzman  i cant see it on boards 
[image]

ch0mik, 2024-10-03:
ok,

i see it now : just ttgo-t-beam-v1_2

[env:ttgo-t-beam-v1_2]
extends = env:esp32
board = ttgo-t-beam
build_flags =
	-Werror -Wall
	-DTTGO_T_Beam_V1_2
	-DHAS_SX1278
	-DHAS_AXP2101
	-DHAS_BT_CLASSIC
	-DHAS_GPS_CTRL
lib_deps =
	${common.lib_deps}
	lewisxhe/XPowersLib @ 0.2.4
	adafruit/Adafruit GFX Library @ 1.11.9
	adafruit/Adafruit SSD1306 @ 2.5.10
	adafruit/Adafruit SH110X @ 2.1.10

richonguzman, 2024-10-03:
yes!