richonguzman/LoRa_APRS_Tracker #162: No Support for Lilygo Supreme V3 433Mhz U-blox MAX-M10S-00B Version?

RSCCal, 2024-10-02:
Hello,
when i try to install the tracker software i always get out, that the message:

"A fatal error occurred: Failed to connect to ESP32-S3: No serial data received."

Im not sure, if this a Issue or is the U-blox MAX-M10S-00B Version unsupported??

Thank you for that project!

richonguzman, 2024-10-02:
hi, did you put the ESP32S3 board (T-Beam Supreme) in boot loader mode? USER + RST ???

RSCCal, 2024-10-02:
Ah! Thanks. Tried this a few times before, but it didn't worked. So i was not sure, is it supported or not. Tried it again, works! Thanks!

richonguzman, 2024-10-02:
test now with the webflasher!!! 

https://richonguzman.github.io/lora-tracker-web-flasher/installer.html

closed 2024-10-02 by richonguzman