richonguzman/LoRa_APRS_Tracker #163: Temperature correction in WebConfig - -Feedback for Webinstaller alpha

DC4FRT, 2024-10-03:
Hi Riccardo,

thank you so much that we can now update our Trackers also with webinstaller and can config this on WebBrowser.

But one small issue:
i need a correction of the temperature (and also the humidity), but the Webconfig does only allow positive values. I need a negative value, e.g. temperature -5.

Some feedback for the Webinstaller:
1. it works on the first flashing ! and that great well, thank you!!!! (my HW SX1268)
2. would it be possible that the tracker can join another WLAN? after 3times pushing the button, not only choose WIFI AP but also choose another WIFI which is already configed
3. may it be possible to change the message-text via WebConfig for the APRSthursday? that would be great !
4. and would it be possible to add a correction for the humidity 

73 DC4FRT, Thomas

DC4FRT, 2024-10-04:
Humidity: it looks also like an offset failure. I have this also on my BME in the iGate. until 10% difference. And yes, i guess that they are o.k. but they have also on humidity an offset failure.

Wifi-connect: to change parameters on Smartphone is o.k. but to do this on my PC with big monitor is more comfortable :-)
Maybe for future: to send also messages over this website, cause i have no KeyBoard and maybe as an alternative to bluetooth connection and an additional app. 

73 Thomas

richonguzman, 2024-10-04:
@DC4FRT please explain me again the last part of the PC connection

so, all need Hum and Temp correction...

DC4FRT, 2024-10-04:
1. change parameters would be more comforable if you can do this on PC when the tracker is in your local network. To start this WLAN connection you can do this as today, after 3times pushing the button then you can choose a.) AP or b.) WLAN Access 
2. feature could also be to send APRS messages over this webside. like message to CA2RXU over the tracker: "Hello how are you" :-) like example: https://www.findu.com/cgi-bin/entermsg.cgi but not send the message over internet but over the tracker if it is in the local Wireless lan. I know, we can do this over KeyBoard (but how i said, many cables and i do not have this). or bluetooth, but there you need an additional app.

richonguzman, 2024-10-07:
temperature now with +-5 correction added 

will take a while to add the hum, because I wanted to add another updates sooner ;) but will return for this

closed 2024-10-07 by richonguzman