richonguzman/LoRa_APRS_Tracker #165: Access Point breakdown V0.9.8

M0IGA, 2024-10-03:
In new firmware with a web configurator, AP doesn't start after save config and rebooting.
Tracker works fine if all parameters were set correctly, but no more AP
Tested on T-Beam v1.2 and ESP32 Wemos lite + E220.

richonguzman, 2024-10-03:
Great then, it was coded this way. Who needs Wifi for regular Tracker use? (besides using it on the car with no problem of power supply?)

richonguzman, 2024-10-03:
does this E220 use LLCC68 ?

M0IGA, 2024-10-03:
> Great then, it was coded this way. Who needs Wifi for regular Tracker use? (besides using it on the car with no problem of power supply?)

Ok, if there is a feature, issue could be closed.
Consider adding an info about it, because I pressed save and have to flash it again. I think it could be helpful for many of us, to avoid mistakes.

M0IGA, 2024-10-03:
> does this E220 use LLCC68 ?

I'll ask, it's not my tracker.

richonguzman, 2024-10-03:
> > does this E220 use LLCC68 ?
> 
> Yes it has.

can you confirm que correct pinout selection of the trackers for it then?

richonguzman, 2024-10-03:
> Ok, if there is a feature, issue could be closed. Consider adding an info about it, because I pressed save and have to flash it again. I think it could be helpful for many of us, to avoid mistakes.

sure thing, I'm writing the WIKI right now with the details to get it more easy

PS: if you need to flash again , just press 3 times and in the menu select Wifi AP again and it will start all over again

closed 2024-10-07 by richonguzman