richonguzman/LoRa_APRS_Tracker #167: Exit Tracker WiFiAP mode

MILUX-DE, 2024-10-08:
Is there only a key combination to enter WiFiAP mode, instead of using e.g. same combination to exit that mode? Rebooting leaves tracker in WiFiAP mode, only method to return to tracker mode seems pressing 'Save' in the WebUI. Adding a 'Reboot to tracker mode' in the WebUI might be useful as well.

MILUX-DE, 2024-10-09:
If you enter WiFiAP mode by mistake just on the device, it should be somehow possible to exit it without any WebUI interaction. Double click might be a logical choice because it has the same meaning when exiting a menu. 

Same applies to the WebUI, if you want to cancel changes instead of "Save", but 'Reboot' exits to WifiAP mode again. An additional choice enabling reboot to tracker mode should be offered.

richonguzman, 2024-10-10:
Now WiFiAP will stop after 1 min of no client connected to the WifiAP

so if started by error, it will return to normal in 1 min.
if started and you dont have any pc o phone to press "save" button it will still return to normal Tracker mode after 1 min

this is added to WebInstaller V2.0.1

closed 2024-10-10 by richonguzman