richonguzman/LoRa_APRS_Tracker #168: Restart without AP signal

BH9FPF, 2024-10-09:
Hello blogger, after flashing the program through the web page. After restarting the device, the AP signal cannot be searched. Can it be fixed in the next version? Have you also considered defining a button function that can be used to switch pages. My device is ttgo-lora32 2.1

richonguzman, 2024-10-09:
The normal behavior is that WifiAP is stopped after webConfiguration is saved and Tracker restarts. As I see the Lora32 v2.1 boards needs a button to access the WifiAP configuration again. Should I add a button in firmware for you to add it as a DIY solution?

BH9FPF, 2024-10-09:
I personally hope that the WifiAP signal remains searchable at all times, making it convenient to adjust configuration information. By adding a button, I mean enabling users to flip through pages and view OLED display information. Of course, these are all my personal suggestions, and the specific implementation still requires comprehensive consideration of the opinions of your other friends.

S57PNX, 2024-10-09:
WifiAP signal doubles the power consumption, so by having it available all the time you reduce the battery autonomy in half. This is not a good design decision.

richonguzman, 2024-10-09:
> WifiAP signal doubles the power consumption, so by having it available all the time you reduce the battery autonomy in half. This is not a good design decision.

this is why I make WiFiAP available only when configuration is needed:
150mA with WiFiAP vs 60-80mA for tracker normal mode.

S57PNX, 2024-10-09:
Yes, my comment was addressed to BH9FPF who requested to have WifiAP available all the time, which is not a good idea.

richonguzman, 2024-10-10:
Now WiFiAP will stop after 1 min of no client connected to the WifiAP

- so if started by error, it will return to normal in 1 min.
- if started and you dont have any pc o phone to press "save" button it will still return to normal Tracker mode after 1 min

richonguzman, 2024-10-10:
this is added to V2.0.1 in WebInstaller

closed 2024-10-10 by richonguzman

BH9FPF, 2024-10-11:
You're amazing. I greatly admire your technical level. I want to learn more from you.

Vic152, 2024-10-27:
Call me stupid guys but where is user button on Liligo TDeck?

richonguzman, 2024-10-27:
USER button is the boot but I forgot to add this as a valid user button, will fix it very soon

Vic152, 2024-10-27:
> USER button is the boot but I forgot to add this as a valid user button, will fix it very soon

so does this mean I have to flip 2 position switch on TDeck very fast 3 times?

richonguzman, 2024-10-27:
> > USER button is the boot but I forgot to add this as a valid user button, will fix it very soon
> 
> so does this mean I have to flip 2 position switch on TDeck very fast 3 times?

no.

i dont know what leads you to this conclusion, but no, sorry: the user button does not work as it should on T-deck (tomorrow I will try to add it)