richonguzman/LoRa_APRS_Tracker #171: Heltec Wireless Tracker looping on startup

pdevlin13, 2024-10-12:
Hi Ricardo
As discussed on the SOTA forum, I have opened the looping as an issue, so that I can share the attached video of the looping.
I have iPhone BLE enabled, I think someone suggested it only happened in this Bluetooth mode but I haven’t tried other BT configs yet.
Happy to do any testing 

Paul MW0PDV 

https://github.com/user-attachments/assets/56958d9e-1829-44e3-a0dd-dfae75e9fccd

richonguzman, 2024-10-12:
Great let's find out what's going on with this:

1) let's try it with latest firmware 2.0.2 please
2) change only the callsigns and nothing else on the webUI 
3) select on tracker menu the freq you need (third one I assume) this will be saved on the internal memory and will start with it all the next times it boots

pdevlin13, 2024-10-12:
ok, fresh install as below, tracker starts up fine.
Reflash and enable iPhone BLE tracker starts looping.

> Great let's find out what's going on with this:
>
> 1. let's try it with latest firmware 2.0.2 please
> 2. change only the callsigns and nothing else on the webUI
> 3. select on tracker menu the freq you [...]

richonguzman, 2024-10-12:
did you change the frequency at all? anything besides only CALLSIGN?

is the callsign saved with any "white/black" space behind the SSID?

pdevlin13, 2024-10-12:
I only changed callsign, and this works no looping.
If I only change callsign and enable iphone BLE it starts looping.

I am trying to do some crude diagnostics, and so far established it returns from the POWER_Utils::batteryManager() call at line 170ish in  Lora_APRS_Tracker_main before looping. 
[image]

pdevlin13, 2024-10-12:
The terminal output from the testing mentioned above
[image]

richonguzman, 2024-10-12:
so is this inside "obtainBatteryInfo()" ???

which board have you? V1 or V1.1?

pdevlin13, 2024-10-12:
> so is this inside "obtainBatteryInfo()" ???

No its inside BatteryManager

[image]

I will check version, just neet to open the enclosure.

richonguzman, 2024-10-12:
> > so is this inside "obtainBatteryInfo()" ???
> 
> No its inside BatteryManager
> 
> <img alt="Screenshot 2024-10-12 152951" width="704" src="https://private-user-images.githubusercontent.com/37481249/375964637-6a4ac301-8ba6-438c-8793-1266b50d90f6.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjg3NDM5MDIsIm5iZiI6MTcyODc0MzYwMiwicGF0aCI6Ii8zNzQ4MTI0OS8zNzU5NjQ2MzctNmE0YWMzMDEtOGJhNi00MzhjLTg3OTMtMTI2NmI1MGQ5MGY2LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMTIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDEyVDE0MzMyMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTBhMTYyYTJiYTY3Y2E5ZDdhOWRmNzc2ZmFiYTU0YjFhNDg2N2E0ZGM0NTRiZDJkZmFhMmRlZjYxNGE3NDk5NWMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.wvklehQweIToqN3jg9KuBIBfBuI-R1H7SB9PQ77G3x0">
> I will check version, just neet to open the enclosure.

yes, but "obtainBatteryInfo()" is the one called there... so it needs to check inside it to know what is not being right

it checks board has ADC_CTRL (which heltecs do have) and also BATTERY_PIN

pdevlin13, 2024-10-12:
Oh yes sorry just notice the call in the if statement.
My version is  1.1

pdevlin13, 2024-10-12:
Ok, this is a bit strange, for each loop  it always seems completes the setup() function but only sometimes gets as far a calling  POWER_Utils::batteryManager()
Hope this helps :-)

richonguzman, 2024-10-12:
will have to change the way this is checked then, give me a few days to get with a test code

pdevlin13, 2024-10-13:
Hi Ricardo
I have done some m ore digging today as the previous testing seemed to produce random results.
It appears to be the pAdvertising->start() call, in BLE_Utils that is causing the Heltec to loop. If I put a sleep 20 after this call it still cycles instantly. Commenting this line out also stops it cycling.

[image]

richonguzman, 2024-10-29:
Hardware test !!!! 

add a 100uF /16V between pin 5 (3.3v) and 4 (GND) as it seems on some trackers the voltage at boot is dropping a small amount of time not allowing full boot process

pdevlin13, 2024-10-29:
> Hardware test !!!!
> 
> add a 100uF /16V between pin 5 (3.3v) and 4 (GND) as it seems on some trackers the voltage at boot is dropping a small amount of time not allowing full boot process

I will give it a try tomorrow 
Thanks

pdevlin13, 2024-10-31:
Hi Ricardo, as in SOTA reflector this does seem to be the solution. I happy to close the issue 😀
Thanks
Paul MW0PDV

richonguzman, 2024-10-31:
for all who will look a this in the future:

HELTEC WIRELESS TRACKER: add 100uF /16V electrolytic capacitor between pin 5 (+) and 4 (- GND) to solve random reboots.

closed 2024-10-31 by richonguzman