richonguzman/LoRa_APRS_Tracker #174: How to connect - configure GPS

pki791, 2024-10-23:
Hi.
Flashed ttgo-lora32-v2.1-gps to the liligo lora32 module, where to connect the GPS module?

I tried pins 34 & 12 but OLED still shows Error of GPS. Any ideas how to proceed? I tried the BN-880 GPS module.

Is there a setting of baud rate, or is there some automation for this programmed?

richonguzman, 2024-10-23:
hi, never added BN-880 gps module to code...  does it need a special baud_speed?

how are you powering it?

pki791, 2024-10-24:
I am powering it from 5V from the board. Which are the correct pins to connect? For now i am not shure what baud the module uses by default, also i not see anywhere where to set the baud in the tracker, or what rate and pins are used?

richonguzman, 2024-10-24:
you have to put some effort on getting the info for this tracker to help me write a code or a modification for you to test.

5V and GND and RX and Tx pins are the ones needed in this tracker?
did you manage to get the baud speed info for it?

richonguzman, 2024-10-24:
you can test this flashing with VSCODE:
go to /src/gps_utils.cpp, line 18 and change it from 9600 to 57600. 
save the file
upload ONLY firmware (the arrow pointing to the right) and it "should" work.

RA3TOX, 2024-10-30:
I have the ttgo-lora32-v2.1
I use pins TX->GPIO34, RX->GPIOI12 and GPS NEO-M8N.
Everything works fine for me

closed 2025-03-18 by richonguzman