richonguzman/LoRa_APRS_Tracker #178: Add support for Heltec Mesh node T114 (Nordic NRF52840)

sysadminpower2019, 2024-11-03:
Hello,

I see a cool new device that has been released. with low power usage and a color screen perfect for ham aprs stuff. Meshtastic already provides support.
here is a link to the aliexpress site. hopefully it is possible to port the firmware to it

https://www.aliexpress.com/item/1005007460963705.html?spm=a2g0o.productlist.main.11.7a751224GpZs8h&algo_pvid=00a82d52-37bf-43a1-90a6-e8153e0ed547&algo_exp_id=00a82d52-37bf-43a1-90a6-e8153e0ed547-5&pdp_npi=4%40dis%21EUR%2153.64%2132.19%21%21%2157.00%2134.20%21%40211b617a17305988765484510eec2c%2112000041287472352%21sea%21DE%21141715380%21X&curPageLogUid=sxiM5bgW0oSv&utparam-url=scene%3Asearch%7Cquery_from%3A

richonguzman, 2024-11-03:
I would love to but I will ask heltec if they can send me one to develop code for it.

sysadminpower2019, 2024-11-08:
y si te pido uno modulo en aliexpress y te lo mando? vi que para CL si funciona bien el envio de la china

richonguzman, 2024-11-08:
> y si te pido uno modulo en aliexpress y te lo mando? vi que para CL si funciona bien el envio de la china

si tienes telegram escribe al mismo usuario que tengo aqui en github y coordinamos ;)

Mike06200, 2025-01-09:
Hello CA2RXU,
If you don't have this board (Heltec T114), give me your address in MP, I send you a new set.
Thanks for your very good work !
Best 73's and happy new year.
Michel / F5AAJ (f5aaj@free.fr)

richonguzman, 2025-01-09:
mail sent :)

closed 2025-01-09 by richonguzman