richonguzman/LoRa_APRS_Tracker #179: No way to set GPS precision

pagex, 2024-11-04:
For privacy reasons it would be good to have setting similar to Meshtastic where you set Precise location to radius of XYZ meters around actual position.

richonguzman, 2024-11-04:
I get you proposal but there are two things to talk about:

1) I can do it as some people would find this useful but...

2) APRS is not encrypted , not hidden, not ment for this. Could be done but the tracking procedure is ment to be somehow "exact" , right?

pagex, 2024-11-04:
Valid points but we are talking only say 50 meters, I have my tracker in car and If I forget to turn it off when coming home it will report it to aprs.fi with exact precision and there is no way to delete once it there. Some similarity can be at Flight tracker sites where they don't report plane positions immediately always add some safety delay / margin. Maybe better way would be not to report position when inside some home point / no report geofence.

richonguzman, 2024-11-04:
> Valid points but we are talking only say 50 meters, I have my tracker in car and If I forget to turn it off when coming home it will report it to aprs.fi with exact precision and there is no way to delete once it there. Some similarity can be at Flight tracker sites where they don't report plane positions immediately always add some safety delay / margin. Maybe better way would be not to report position when inside some home point / no report geofence. BTW, Thanks for your work on that project !

maybe a rule should be created for having a specific "error" near certain places (custom places for each callsign?)

or just make everyone be more more responsible for turning it off if needed.

closed 2025-03-18 by richonguzman

sjmurdoch, 2025-08-24:
Was this or equivalent implemented? I couldn’t find out how to configure privacy settings, but perhaps I am missing something. Thanks!

richonguzman, 2025-08-24:
> Was this or equivalent implemented? I couldn’t find out how to configure privacy settings, but perhaps I am missing something. Thanks!

hi!

this will be on next update ;)

richonguzman, 2026-01-13:
> Hi [@richonguzman](https://github.com/richonguzman), has this been implemented in [V2.4](https://github.com/richonguzman/LoRa_APRS_Tracker/releases/tag/V2.4)? Many thanks for your hard work!

Hi, this has not been added into the last Release (v2.4) I see it was not that urgent to add for a tracker the position ambiguity as it was for the Digipeater/iGate.

do you still need it fast enough to create a minimal new release (like V2.4.1)

usefulnoise, 2026-01-13:
> Hi, this has not been added into the last Release (v2.4) I see it was not that urgent to add for a tracker the position ambiguity as it was for the Digipeater/iGate.
> 
> do you still need it fast enough to create a minimal new release (like V2.4.1)

Thanks for your feedbak @richonguzman!
For my personal use case, having this feature in my tracker would be very useful 🚀