richonguzman/LoRa_APRS_Tracker #180: ESP32 + LoRa E22 + GPS

RA3TOX, 2024-11-08:
I downloaded the latest version from October 29.
After configuring via the address 192.168.1.4 (I entered only the call sign) the module does not work and does not respond. What could be the problem?

richonguzman, 2024-11-08:
did it reboots and creates WiFiAP, right?

did you connect the pins just as the /src/boards_pinout.h instructions for this board?

which modules do you have as E22 and GPS ?

RA3TOX, 2024-11-09:
I didn't create a point WiFiAP. 
Is this a requirement?
I entered only my call sign

I connect the pins just as the /src/boards_pinout.h instructions for my board

I'm using modules E22-400M30S and NEO-M8M

richonguzman, 2025-01-01:
hi, did you get this solved or should we continue?

closed 2025-03-18 by richonguzman