richonguzman/LoRa_APRS_Tracker #181: How to troubleshoot BT not being seen in an Android tablet or phone?

GFaza16550, 2024-11-20:
Hi Ricardo!

Como andás? He configurado dos Heltec v3 como iGate y funcionan bien.
Y he configurado dos Heltec Wireless Tracker como Tracker y funcionan bien

Pero, quise volver a configurar uno de los v3 como TNC, dado que quiero recibir los beacons en una tablet Android.
Y por más que lo configuro como BT Clásico, no lo puedo ver en la lista de dispositivos.
Si lo pongo como BLE lo puedo ver, pero ya sabemos que APRSDroid no anda con dispositivos BLE.

Que sugieres? No logro darme cuenta como verificarlo.

Gracias.

GFaza16550, 2024-11-20:
And, just to add a comment, i flashed that v3 with the latest Meshtastic firmware, and BT worked fine.
I flashed again the v3 with your web flasher, and an android tablet and also a Samsung A55 android phone are not able to see the device.

Also, i activated the switch from the configuration in the v3, and it didn't work.

richonguzman, 2024-11-20:
Hi

Heltec v3 and wireless tracker are all esp32s3. This processor only does BLE (and has not BT classic)

Meshtastic app works with all types of BT (classic and BLE).

A proposal was sended to APRSdroid to work with BLE (and on the beta code it works with BLE) but right now they have not updated it, so it only works with BT Classic (for now)

GFaza16550, 2024-11-20:
Thanks!
I forgot that!
Thanks!

GFaza16550, 2024-11-20:
How do we get that ble beta version of APRSDroid? I didn't find it in their website.

richonguzman, 2024-11-20:
> How do we get that ble beta version of APRSDroid? I didn't find it in their website.

will try to get it to share...