richonguzman/LoRa_APRS_Tracker #185: TWR Plus Project

sysadminpower2019, 2024-11-27:
Hello,

I read on the lilygo site that you were building an APRS tracker project with the TWR plus but I can't seem to find any info on github and couldn't find a better place to ask about it.

thanks

richonguzman, 2024-11-27:
I have this almost ready:

Works as a handy , has gps , screen and sends tones 

But I can't find q way to send afsk and decode it

sysadminpower2019, 2024-11-27:
there are a few projects that use this same or similar hardware layout maybe you can get some inspiration from them 

https://github.com/nakhonthai/ESP32APRS_T-TWR
https://github.com/Hamlabs/ArcticTracker-ESP32
https://github.com/VanceVagell/kv4p-ht
https://github.com/OpenRTX/OpenRTX/tree/master/platform/targets/ttwrplus

plus here is the firmware that I had to install on my openedition twr to get it to work

https://github.com/OpenRTX/sa8x8-fw

S57PNX, 2024-11-27:
Note that my T-TWR spontaneously went into short circuit, heated up and started smoking while sitting on my shelf in off state and untouched for more than a week. Lilygo claim it is impossible, but I am sure if by lucky coincidence wasn't in the room at the time it would have caused a fire in my house.

richonguzman, 2024-11-27:
> there are a few projects that use this same or similar hardware layout maybe you can get some inspiration from them
>
> https://github.com/nakhonthai/ESP32APRS_T-TWR [...]

sadly most of them dont add no real info how to do things...

richonguzman, 2024-11-27:
> Note that my T-TWR spontaneously went into short circuit, heated up and started smoking while sitting on my shelf in off state and untouched for more than a week. Lilygo claim it is impossible, but [...]

wow! scary

sysadminpower2019, 2024-12-05:
quizas puedas hablar con este chico 

https://github.com/VanceVagell/kv4p-ht/issues/90#event-15512053536

sysadminpower2019, 2024-12-15:
estoy es lo que me respondio

I'm using the library javAX25 which does the AFSK modulation and
demodulation for me. Maybe they could look at that and see what they can
reuse from it?

sysadminpower2019, 2025-01-01:
maybe you could ask the developer from the other project if you could over it. it's been almost a year without any updates and he doesnt respond anymore.

https://github.com/nakhonthai/ESP32APRS_T-TWR

richonguzman, 2025-01-01:
> maybe you could ask the developer from the other project if you could over it. it's been almost a year without any updates and he doesnt respond anymore.
> 
> https://github.com/nakhonthai/ESP32APRS_T-TWR

he answers , but this code is really nothing friendly to get it working

but anyways I will get this working this next month for sure (I want it too)

sysadminpower2019, 2025-01-02:
thank you. i will take a look also at providing you a bit of econonic help via github. your software now basically powers half the aprs station around the world. 

thanks for the amazing job

sysadminpower2019, 2025-02-04:
Hello,
Just wanted to check up if you had any time to look into the TWR project. I know you are a very busy and I thank you for your constant work and support of the community

richonguzman, 2025-02-05:
Hola,

as soon as I finish a project about sending files over lora I will return to it, hang in there, soon!

richonguzman, 2025-03-18:
i have not forgotten this!

I'm still fighting to get it working

sysadminpower2019, 2025-05-23:
Hola.

Como te va con este proyecto?

richonguzman, 2025-05-23:
> Hola.
> 
> Como te va con este proyecto?

no sabes las veces que he ido peleando con esto... a veces logro algo... pero nada listo aun.

me tiene super frustrado de tantos intentos, pero no me dare por vencido y pronto volvere a por ello.

miodzicho, 2025-06-04:
Any progress ? nakhonthai did really great job, with GUI etc. But last version doesn't work...and dev is not active it seems anymore.

richonguzman, 2025-06-04:
> Any progress ? nakhonthai did really great job, with GUI etc. But last version doesn't work...and dev is not active it seems anymore.

once i finish a (soon to come) LoRa iGate big update i will return to this and also Tracker updates.