richonguzman/LoRa_APRS_Tracker #188: LILYGO T-Beam 1.2 with new PMU AXP2101

OpaCode, 2024-12-09:
I bought a T-Beam 1.2 with AXP2101. When compiling with VisualStudioCode, however, I get the warning that firmware.map is not found and therefore the error message: 
“collect2.exe: error: ld returned 1 exit status
*** [.pio\build\ttgo-t-beam-v1_2\firmware.elf] Error 1”. 

The board is in the list of supported devices. What could I have done wrong? I can flash other FW without errors.

richonguzman, 2024-12-09:
Please let me know which OS do you have? which version of platformio do you have? have you imported it or just opened it?

did you try just flashing with the webflasher?

OpaCode, 2024-12-09:
Thanks for pointing out the Webflasher. I found it and was able to install the firmware with it. 
But it does not work with VisualStudioCode. I have Windows 11 and the PlatformIO IDE v3.3.3. I have tried both options. It does not work with import and also not with open folder.

closed 2024-12-09 by OpaCode

richonguzman, 2024-12-09:
> Thanks for pointing out the Webflasher. I found it and was able to install the firmware with it. But it does not work with VisualStudioCode. I have Windows 11 and the PlatformIO IDE v3.3.3. I have tried both options. It does not work with import and also not with open folder.

I guess this is a local issue with the opening or importing projects. Here it runs good ;)

glad you could install it with webflasher