richonguzman/LoRa_APRS_Tracker #189: T-Deck Plus keyboard/joystick issues

MILUX-DE, 2024-12-17:
Trying to navigate through menues is actually inaccurate with the joystick, it needs more finetuning because it actually triggers not intended actions, even sometimes leading to a total reset. Using the center button for actions requiring Long or Double Press often won't work, too.

Some keys (as well useful for other keyboards without joystick) should be assigned additionally, e.g. ENTER, BACKSPACE and W-A-S-D for up-left-down-right to ease navigation (outside any text entry mode).

Last not least, Message Write selection on T-Deck Plus pretends: "No Keyboard Detected. Can't write Message" 🙁

richonguzman, 2024-12-18:
Hi,

in two or three days I will update Tracker Firmware with upgrades for T-deck and joystick, keyboard and experience y way better now.

MILUX-DE, 2024-12-22:
Joystick action has improved, but still: "No Keyboard Detected. Can't write Message" 

Log Setup:
[INFO][Main] RichonGuzman (CA2RXU) --> LoRa APRS Tracker/Station.
[INFO][Main] Version: 2024.12.21.
[INFO][LoRa] Initializing SX126X ....
[INFO][LoRa] LoRa init done!.
[INFO][Main] T-Deck Keyboard Connected to I2C.
[INFO][Main] Touch Module Connected to I2C.
[INFO][Main] Setup Done!.

followed by I²C errors:
[ 16209][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[ 19300][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[ 22388][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[ 25333][E][Wire.cpp:513] requestFrom(): i2cRead returned Error -1
[ 25421][E][Wire.cpp:513] requestFrom(): i2cRead returned Error -1
...

xtension2025, 2025-01-01:
I have same issue, but sometimes the
 Keyboard was working, sometimes not working, if it's working, press the "enter" key can wake up the screen,if not, nothing happened

richonguzman, 2025-01-01:
will try to check it out as latest firmware should have solve this issue finding the Keyboard. 

T-deck is a good product/board but sometimes it gets confused and does not read fine the Keyboard

xtension2025, 2025-01-01:
My t-deck plus with Firmware v2.1.1

richonguzman, 2025-01-02:
could you test it a little further? v2.1.2 is up

xtension2025, 2025-01-03:
I am flashed 2.1.2, but don't select "erase", test it and found the keyboard was still not stable, should I select "erase" to flash ?

richonguzman, 2025-01-03:
erase or not would only change the saved data on internal memory (SPIFFS) of your configuration. So it wont change the way it works.

sadly T-deck does this thing that it wont read Keyboard , but after a reset or pressing the keyboard it will be recognized.

test this "bad" solutions to see if yours work (and if you find any solution or testing with new addresses for I2C let me know)

MILUX-DE, 2025-02-26:
Even release V2.2.2 brings no improvement for T-Deck plus keyboard use, e.g. in Messsage Mode "No Keyboard Detected. Can't write Message" despite the boot log entry "T-Deck Keyboard Connected to I2C". Touchscreen and trackball are working, but keys are ignored. No change after reset or pressing keyboard. 

When flashing the same device with Meshtastic or other software, there are no problems at all using the keyboard. Maybe looking at other source repositories might give a hint why it's not working here?

MILUX-DE, 2025-04-10:
I don't think it's an adress issue, because keyboard is recognized during boot sequence, but get's "lost" later on. Maybe it's a timing issue? Meshtastic or Meshcom on the same device show no keyboard issues.

richonguzman, 2025-04-10:
> I don't think it's an adress issue, because keyboard is recognized during boot sequence, but get's "lost" later on. Maybe it's a timing issue? Meshtastic or Meshcom on the same device show no keyboard issues.

and what do they do to solve this?