richonguzman/LoRa_APRS_Tracker #192: Confusing rssi values

stwendl, 2024-12-25:
I run your latest code (dec 6, 24) on a heltec v3 and tlora v1.6.1  I am observing some rather odd results in the reported rssi values as displayed in the web and syslog data.  For instance, a constant remote station may report -30 as rssi using a stubby antenna that came with the board and is probably not the best choice.  Connecting a better antenna (5.8dbi gain) results in a rssi value -129. Per definition of rssi, a worse signal.

according to this article https://forum.arduino.cc/t/negative-rssi-issue-with-esp32-and-lora-ra-02-sx1278-module/1248199/9 the real rssi value should be calculated for the lf port like

Rssi(dbm) = -164 + (Lora rssi)

so in my case that would add with the stubby antenna being -164 + 30 =-134  dbm mor the better antenna -164 + 129 =-35 dbm

Any thoughts

stwendl, 2024-12-25:
That report references the sx1278 chip and the heltec uses sx1262. Assuming it may be afflicted by the same assumption

stwendl, 2024-12-25:
Also looking at the code include files, the rssi value is defined as "int" and could not be negative. Whereas the snr is defined as float and supports negative values. If these definitions are from Lora designs then this would suggest those formulas above being meaningful

richonguzman, 2024-12-26:
Thanks for pointing this out

I will try to validate all of this real soon (29 dec as I return home)