richonguzman/LoRa_APRS_Tracker #203: wiki -> 103.-BME280-BMP280-BME680-Si7021.md

chansizzle, 2024-12-31:
update text for clarity. Too many pronouns created ambiguity:

BME280 (BMP280, BME680, Si7021) sensor modules can be added by connecting them as I2C (parallel to the Oled Screen).

The Firmware has code to look for the sensor modules and their I2C addresses, for you so just connect the sensor module and the firmware will find sensor module and work with it.

To see this data on the Oled Screen or transmit it as Wx telemetry, activate it in the *tracker_conf.json* file.

For example: SCL (IO22) and to SDA (IO21) on the LoRa32 board. Look at **/src/boards_pinout.h** for the specific pins for I2C on your board.

Wx data will be sent once 10 min as APRS.fi does not process faster Wx APRS packet.

chansizzle, 2024-12-31:
What exactly does this mean?

> Wx data will be sent once 10 min as APRS.fi does not process faster Wx APRS packet.

Does this mean
A. the Wx data will be sent every 10 minutes?
or
B.  the Wx data will be sent one time after 10 minutes of time since boot?
C. Or some other meaning?
I suspect it is the case of A

richonguzman, 2024-12-31:
Added, 

Wx data is sent once each 10 min and no sooner as APRS.fi won't process any packet with WX data that it receives sooner than 10 min between packets

chansizzle, 2024-12-31:
Newest in wiki:

Wx data will be sent once 10 min as APRS.fi does not process faster Wx APRS packet.

Small change to:

Wx data will be sent once every 10 minutes as APRS.fi does not process faster Wx APRS packets.

richonguzman, 2024-12-31:
> Newest in wiki:
> 
> Wx data will be sent once 10 min as APRS.fi does not process faster Wx APRS packet.
> 
> Small change to:
> 
> Wx data will be sent once every 10 minutes as APRS.fi does not process faster Wx APRS packets.

I used this:

Wx data will be sent once every 10 minutes as APRS.fi does not process Wx APRS packets at a faster rate.

chansizzle, 2024-12-31:
:thumbsup: :ok_hand:
good clarity

closed 2025-01-01 by chansizzle

richonguzman, 2025-01-01:
thanks for the help

if more ideas comes to you, write me on Telegram app with the same user as I have here on github