richonguzman/LoRa_APRS_Tracker #207: T-Deck + GPS (DIY) no display except for symbol

S57PNX, 2025-01-02:
I've upgraded the original T-deck + GPS (DIY) from 26.6.2024 to 21.12.2024. Flashing and webconfig went OK, device boots normally according to logs but the display remains black. However, the little man symbol does show on the display!

It looks like the display is correctly initialized, just the display text is not sent to it and only the symbol is sent. 

[10:35:01]ESP-ROM:esp32s3-20210327
[10:35:01]Build:Mar 27 2021
[10:35:01]rst:0x15 (USB_UART_CHIP_RESET),boot:0x8 (SPI_FAST_FLASH_BOOT)
[10:35:01]Saved PC:0x40379d96
[10:35:01]SPIWP:0xee
[10:35:01]mode:DIO, clock div:2
[10:35:01]load:0x3fce3808,len:0x44c
[10:35:01]load:0x403c9700,len:0xbe4
[10:35:01]load:0x403cc700,len:0x2a38
[10:35:01]entry 0x403c98d4
[10:35:01]E (358) psram: PSRAM ID read error: 0x00ffffff
[10:35:07][INFO][Main] RichonGuzman (CA2RXU) --> LoRa APRS Tracker/Station
[10:35:07][INFO][Main] Version: 2024.12.21
[10:35:07][INFO][LoRa] Initializing SX126X ...
[10:35:07][INFO][LoRa] LoRa init done!
[10:35:10][INFO][Main] Touch Module Connected to I2C
[10:35:12][INFO][Main] Setup Done!

[image]

S57PNX, 2025-01-02:
If I install the T-deck Plus firmware, the display works as expected. Should I run this firmware? I don't see any difference between these two platforms...

richonguzman, 2025-01-02:
hi, can you please update to the latest firmware available and confirm it does show as expected? if not let me know to dive a little and see what's the problem

S57PNX, 2025-01-02:
Hey Richon, I am already on the latest FW 21.12.2024. I am sure that you are missing || defined(TTGO_T_DECK_GPS) somewhere in TFT code, but my development machine is broken and I can't try to find it.

S57PNX, 2025-01-02:
Maybe line 492 in display.cpp?

void displayShow(const String& header, const String& line1, const String& line2, const String& line3, const String& line4, const String& line5, int wait) {
    #ifdef HAS_TFT
        #if defined(TTGO_T_DECK_PLUS)
            draw_T_DECK_Top();//header, line1, line2);
            draw_T_DECK_Body(header, line1, line2, line3, line4, line5);

richonguzman, 2025-01-02:
> Maybe line 492 in display.cpp?
>
> void displayShow(const String& header, const String& line1, const String& line2, const String& line3, const String& line4, const String& line5, int wait) { #ifdef [...]

can you please add the missing line : " || defined(TTGO_T_DECK_GPS)" 

as it should be "almost" the same that the T-DECK Plus code? to test?

richonguzman, 2025-01-02:
just checked, this is the problem :(

S57PNX, 2025-01-02:
Correct, this was the problem. Adding the defined fixed it.

richonguzman, 2025-01-02:
> Correct, this was the problem. Adding the defined fixed it.

great

will correct it right away and add to webflasher in less than an hour

richonguzman, 2025-01-02:
now into webflasher v2.1.2

closed 2025-01-02 by richonguzman