richonguzman/LoRa_APRS_Tracker #209: Lilygo LORA32 TTGO V2 (V1.6) reenter web ui

TheJimmyNowak, 2025-01-02:
Hi,
I'm experiencing an issue while trying to re-enter the web UI. My TTGO LoRa32 V2(1.6) board doesn't have a user button. I followed the suggestions mentioned in [this thread](https://meshtastic.discourse.group/t/ttgo-lora32-v2-v1-6-button/522/6).

The method works on a board running Meshtastic, but it doesn't work on a board running APRS. Specifically, attempting to push the button three times to re-enter the web UI fails.

richonguzman, 2025-01-02:
I've just added PIN15 for this... you can add a button there and it should work 

let me know how it goes with webflasher V2.1.2 which has this update

TheJimmyNowak, 2025-01-02:
Thank you for your fast answer! It works. 73 de SP8UWU

closed 2025-01-02 by richonguzman