richonguzman/LoRa_APRS_Tracker #214: T-BEAM AXP2101 start on battery

lemielek, 2025-01-04:
Hi and TNX.
Few days ago I have flashed with web flasher (december version) my T-Beam v.1.2 with AXP2101 and it was working on USB cable.
Then I disconnected cable and few hours later I pt m battery in it, but it doesn't start. Changed battery, also no start.
I connected cable with 5V and it worked again. Disconnected cable and it still works.
So I get battery off it, put in again and it does not start. After connecting cable it started so I took cable off.

Another interesting thing was that I put in it battery not freshly recharged, but in good condition.
When in started battery level percentage started groving from zero to 3%. I am sorry, I did not wait more and connect USB cable to the charger until it was fully charged. 
Now I switched battery to another and this one shows 97% so maybe that one before was somehow tricky.

And again, I see that there is new version from 2 January, should I reflash to new version and check again? I do not see anything related in last commits.

richonguzman, 2025-01-07:
I would need to re-flash it again with webflasher and check if the boards loads the first AXP2101 in serial output info. If not there should be more testing to do.

lemielek, 2025-01-07:
T-BEAM v1.1. turns on when battery is connected.
[18:14:51]ets Jul 29 2019 12:21:46
[18:14:51]
[18:14:51]rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[18:14:51]configsip: 0, SPIWP:0xee
[18:14:51]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[18:14:51]mode:DIO, clock div:2
[18:14:51]load:0x3fff0030,len:1184
[18:14:51]load:0x40078000,len:13232
[18:14:51]load:0x40080400,len:3028
[18:14:51]entry 0x400805e4
[18:14:53][INFO][AXP192] init done!
[18:14:57][INFO][Main] RichonGuzman (CA2RXU) --> LoRa APRS Tracker/Station
[18:14:57][INFO][Main] Version: 2025.01.02
[18:14:57][INFO][LoRa] Initializing SX127X ...
[18:14:57][INFO][LoRa] LoRa init done!
[18:14:58][INFO][Main] Setup Done!

lemielek, 2025-01-07:
Yes.
Sorry, my stupidity. I did not noticed that PWR button works.
v1.2 on battery starts wilt long pressed PWR button. v.1.1 starts automatically.
I will close that,

closed 2025-01-07 by lemielek

richonguzman, 2025-01-07:
now you know :)

good that this is solved, enjoy!