richonguzman/LoRa_APRS_Tracker #215: Draft - T-Deck + GPS (DIY) - no GPS

Vic152, 2025-01-06:
This is in DRAFT since I do not now if FW would effect GPS in any way. There was report from @madpsy_ that on his T-Deck Plus GPS only worked after second flash.

GPS: HGLRC M100 MINI M10 GPS
T-Deck: Original T-Deck 433
Firmware: V2.1.2

Repro: Flash T-Deck with firmware V2.1.2. leave it for several hours to find satellites.

Expected: After some time GPS has a lock and coordinates show on the tracker screen.
Observed: No GPS lock. Coordinates are zeroes.

GPS Info
[image]

richonguzman, 2025-01-07:
Leave it outside

If date and time is updated , fix will get eventually 

It no update in date and time there is need to do some mods

aymericmo, 2025-01-07:
What is your GPS Antenna ? First fix need a lot of clearance.
If you try at the edge of a window with 50% sky hiding it lasts a very long time.
The default antennas provided are not very efficient, I myself have experienced similar problems.

S57PNX, 2025-01-07:
Date 1970-1-1 in your screenshot is a good indicator that the serial communication to the GPS module is not working. My GPS starts with 2000-1-1 before getting a lock, and it takes approx 90 seconds to get a lock on my window which is a bad spot. Outside should be even better.

T-Deck DIY in version 2.1.2 has GPS_BAUDRATE 38400 by mistake, the next versions will again have 9600. If your GPS module really has 115200, that will require a manual change and building own firmware. I suggest to try with a custom firmware build, first try with 9600 and then with 115200. 

See also #211

Vic152, 2025-01-07:
@S57PNX so this worked fine with previous (black and white) version of the FW. I have ability to turn GPS off physically. When I do that it tells me the frames are not detected for what it's worth. As I said I made this bug a draft cause I do not see GPS flashing red either, meaning there is no GPS lock. I will further investigate will open the unit and see if there are any physical damages and confirm in here.

richonguzman, 2025-01-07:
@Vic152 but if you cant wait just upload the firmare with the code from github and vscode

V2.1.3 will be soon updated (one or two days at most)

Vic152, 2025-01-07:
> @Vic152 but if you cant wait just upload the firmare with the code from github and vscode
> 
> V2.1.3 will be soon updated (one or two days at most)

There is no rush at all. This is all for betterment of the community :)

richonguzman, 2025-01-07:
its now added, could not wait for tomorrow :)... please test

Vic152, 2025-01-07:
[image]
Welcome to Y2k

Vic152, 2025-01-07:
[image]
Welcome to continental Europe 🇪🇺

richonguzman, 2025-01-07:
great, GPS works ;)

Vic152, 2025-01-07:
Thank you and let me close it now.

Vic152, 2025-01-07:
.

closed 2025-01-07 by Vic152