richonguzman/LoRa_APRS_Tracker #216: Heltec V3 stuck in boot

InspiringCode, 2025-01-07:
I have tried to flash two different Heltec Lora Wifi 32 V3 devices (`HTIT-WB32LAF`) multiple times with web flasher and with VS Code (I have choosen `Heltec V3 (TNC)` . But after flashing the device doesn't but up, the orange LED blinks and I get the following logs:

```
[20:36:01]ESP-ROM:esp32s3-20210327
[20:36:01]Build:Mar 27 2021
[20:36:01]rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
[20:36:01]SPIWP:0xee
[20:36:01]mode:DIO, clock div:1
[20:36:01]load:0x3fce3808,len:0x4bc
[20:36:01]load:0x403c9700,len:0xbd8
[20:36:01]load:0x403cc700,len:0x2a0c
[20:36:01]entry 0x403c98d0
```

Strangely enough I was able to get into the Web config UI but after saving the config I got an error (`Failed to read config`) and the Heltec never booted up again after that.

I was so looking forward to try this great project but after many hours of trying to get it working I absolutely failed (even though I am a programmer myself). Please help me out!

Here is my config in case it matters:

```json
{
	"beacons": [
		{
			"callsign": "OE6DBR-2",
			"gpsEcoMode": false,
			"symbol": "-",
			"overlay": "/",
			"micE": "",
			"comment": "",
			"smartBeaconActive": false,
			"smartBeaconSetting": 0	
		},
		{
			"callsign": "OE6DBR-2",
			"gpsEcoMode": false,
			"symbol": "[",
			"overlay": "/",
			"micE": "",
			"comment": "",
			"smartBeaconActive": true,
			"smartBeaconSetting": 0
		},
		{
			"callsign": "OE6DBR-2",
			"gpsEcoMode": false,
			"symbol": ">",
			"overlay": "/",
			"micE": "",
			"comment": "",
			"smartBeaconActive": true,
			"smartBeaconSetting": 2
		}
	],
	"display": {
		"showSymbol": true,
		"ecoMode": false,
		"timeout": 4,
		"turn180" : false
    },
	"battery": {
		"sendVoltage": false,
		"voltageAsTelemetry": true,
		"sendVoltageAlways": false,
		"monitorVoltage": true,
		"sleepVoltage": 2.9
	},
	"other": {
		"simplifiedTrackerMode": false,
		"sendCommentAfterXBeacons": 10,
		"path": "WIDE1-1",
		"nonSmartBeaconRate": 15,
		"rememberStationTime": 30,
		"standingUpdateTime": 15,
		"sendAltitude": true,
		"disableGPS": false,
		"acceptOwnFrameFromTNC": false
	},
	"winlink": {
		"password": "ABCDEF"
	},
	"wxsensor": {
		"active": false,
		"temperatureCorrection": 0.0,
		"sendTelemetry": false
	},
	"notification": {
		"ledTx": false,
		"ledTxPin": 13,
		"ledMessage": false,
		"ledMessagePin": 2,
		"ledFlashlight": false,
		"ledFlashlightPin": 14,
		"buzzerActive": false,
		"buzzerPinTone": 33,
		"buzzerPinVcc": 25,
		"bootUpBeep": false,
		"txBeep": false,
		"messageRxBeep": false,
		"stationBeep": false,
		"lowBatteryBeep": false,
		"shutDownBeep": false
	},
	"lora": [
		{
			"frequency": 433775000,
			"spreadingFactor": 12,
			"signalBandwidth": 125000,
			"codingRate4": 5,
			"power": 20
		},
		{
			"frequency": 434855000,
			"spreadingFactor": 9,
			"signalBandwidth": 125000,
			"codingRate4": 7,
			"power": 20
		},
		{
			"frequency": 439912500,
			"spreadingFactor": 12,
			"signalBandwidth": 125000,
			"codingRate4": 5,
			"power": 20
		}
	],
	"bluetooth": {
		"type": 2,
		"active": false
	},
	"pttTrigger": {
		"active": false,
		"io_pin": 4,
		"preDelay": 0,
		"postDelay": 0,
		"reverse": false
	},
	"wifiAP": {
		"active": false,
		"password": "1234567890"
	}
}
```

richonguzman, 2025-01-07:
Hi, 

lets start by validating a few things:
- board is heltec v3 with sx1262 lora module? which freq?
- do you have any battery attached to the board ? (can you test without the battery monitoring enabled?)
- heltec v3 only has BLE (so please disable BT at all if it was enabled at any testing scenario)

InspiringCode, 2025-01-08:
This is [the link](https://www.aliexpress.com/item/1005007752194012.html) to the exact device I have (I ordered version "433MHz shell"). It's a Heltec V3 (V3.2) with SX1262. I have attached some images below. I have no battery attached (enabling the battery monitoring was a really stupid oversight on my side, I am sorry for that, but unfortunately it wasn't the problem).

I tried to flash with stock config today (just putting in my callsign), but same behavior: I can access the access point after web flash, but once I hit save, the device is not reachable anymore, screen blank and orange LED blinks. Tried with Bluetooth BLE enabled and disabled, no difference.

[image]
[image]
[image]

richonguzman, 2025-01-08:
crazy thought:

can you test it connecting the battery to it ?

InspiringCode, 2025-01-08:
Unfortunately I have no battery available :(

richonguzman, 2025-01-08:
> Unfortunately I have no battery available :(

maybe connect (+) side one the battery connector to 3.3V pin... just to be sure its not expecting a battery voltage and (as you dont have any connected) it is powering it off .

richonguzman, 2025-01-08:
as you know VSCODE, we could try many things to isolate the boot up process

do you have Telegram?  find me with the same user as here on github

richonguzman, 2025-01-09:
this will be added tonight

closed 2025-01-09 by richonguzman