richonguzman/LoRa_APRS_Tracker #218: Build issue

snag1es, 2025-01-09:
Unable to build the firmware using the vsCode.

The board I want to use is Lilygo Supreme. Before the platformio.ini contains all system boards, you can launch the board you want to flash. Currently, only t-beam v1.2 is applicable 

Thank you in advance.

richonguzman, 2025-01-09:
Right now works the same as before, only that the boards are put as "variants"

if you really need VSCODE over the Webflasher, check how to enable the environment for your board in VSCODE (not that hard, just press on the lower bottom of the screen over DEFAULT: Tbeam-v1_2 and a new window will open to select the Supreme

snag1es, 2025-01-12:
Thanks, 
All done. I found the extra folder containing all variants; the only need is to change the env type in the main platformio.ini where the variants folder is included as mentioned.

Thanks one more time.

closed 2025-01-12 by richonguzman